raomin/ESPAltherma #231: Code runs on Raspberry

ap20132022, 2023-01-09:
can you image that this code also runs on raspberry? 
The reasons for raspberry:
- I have already my volkszaehler (which runs with smart meter) nearby
- in my cellar Wifi works not quite well
- you can use CAN interface with extension
What do you think about this suggest?

Tardymo, 2023-01-10:
Do you mean the sigle board computer or Pico W?
Communication is done through serial port, so it is possible to port it to any hardware having UART, but it would then be a completely different project.

raomin, 2023-01-10:
Thanks @Tardymo. That's correct, code would have to be ported to Linux but this would not be very difficult...

ap20132022, 2023-01-10:
> Do you mean the sigle board computer or Pico W? Communication is done through serial port, so it is possible to port it to any hardware having UART, but it would then be a completely different project.

I mean single board computer. 
Maybe a new project named RaspiTherma?

raomin, 2023-01-11:
I'll convert this one to a discussion as it's not really an ESPAltherma issue.