raomin/ESPAltherma #234: Home Assistant - Integration sg is not supported

dg1yds, 2023-01-21:
Hello,

I am just "playing" with ESPAltherma, my M5 Stick and PlatformIO.
I am still waiting for the delivery of the heat pump, but I'll play something with the tool in advance.
The configuration and flashing worked without any problems - thank you in advance for the tool and your work.

I would like to use the smart grid function and have configured it accordingly.
I am using Home Assistant 2023.1.6 and the integration of ESPAltherma via mqtt is generally working, although the ESP is not connected to the heat pump. I can see internal values via sensor.althermasensors.

But I think the smart grid integration is somehow "wrong" with my HA version; I can see the following warning in the log; additionally I have added a screenshot of the MQTT data.
A sensor for the smart grid functionally is not available in HA.

**HA log**
"WARNING (MainThread) [homeassistant.components.mqtt.discovery] Integration sg is not supported"

[image]

I am not sure if this is a bug or an issue with my configuration or environment; perhaps you can check.

colseven, 2023-02-06:
Same here. I use SG Switch with this Code in configuration.yaml

`mqtt:      
  switch:
    - name: "SG Altherma"
      state_topic: "espaltherma/sg/state"
      command_topic: "espaltherma/sg/set"
      payload_on: "2"
      payload_off: "0"`

[image]

dg1yds, 2023-02-06:
Hi,
I am not a programmer but I have modified the original code to provide a selector within HA.
I have also carried over some ideas from another open PR (https://github.com/raomin/ESPAltherma/pull/110)

You can try https://github.com/dg1yds/ESPAltherma.
I have merged hopefully all my stuff to the repo.

Give it a try; I can setup an official PR afterwards.

PR #261 (2023-03-12): FIX: Smart grid operating mode MQTT discovery.
Its showing now as a select entity on HA, on the same device as the rest of the entities. The payload needed to trigger each mode is unchanged, the conversion is handled by ha using templates.

ToDo: (in a new PR) S4S (energy meter pulse output) signal generation, so smart grid is fully operational.

CLOSES: #234 

The S4S signal (energy pulse meter otput) IS NOT implemented in this PR. It will implemented in a separated PR
closed 2023-03-12 by raomin