raomin/ESPAltherma #238: Altherma EHSX16P50BB using S protocol?

amasciotta, 2023-01-27:
I own an Altherma EHSX16P50BB, and I've deployed ESPAltherma on a M5StickC Plus.

Initially I tried the `ALTHERMA(HPSU6_ULTRA).h` definition file, but the display on the M5Stick showed `Error 0x15 0xEA returned from HP`. Then I tried the `PROTOCOL_S_ROTEX.h` as suggested in the guides. It seems to work, but I still need to dive inside the meaning of the values I'm seeings.

However I'm frankly disappointed by the fact that my supposedly recent HP implements the old S protocol, with such a limited list of parameters. My HP was installed in 2019.

Am I missing something? Why does my relatively new HP only speaks the old S protocol?

I paste here a list of values received from ESPAltherma using the S protocol.

```
Circulation pump: 'OFF'
External heater?: 'OFF'
Priority to domestic water: 'OFF'
Burner inhibit from solaris: 'OFF'
Refrig. Temp. liquid side(C): 24.625
Inlet water temp.(C): 25.8906
Outlet Water Temp.(C): 25.7969
D(C): 25.8906
DHW tank temp.(C): 47.9844
F(C): 0.00390625
Delta-Tr(deg): 28
R/C Setpoint(C): 0
Operation Mode: Heating
Error Code: ' 0'
Thermo Off Error: ' 0'
Warning Code: ' 0'
Caution Code: ' 0'
'????': 120.941
M5VIN: 4.81V
M5AmpIn: 140mA
M5BatV: 3.96V
M5BatCur: 85mA
M5BatPwr: 0mW
WifiRSSI: '-80dBm'
```

I also attach the photos of my provisional setup and of the plate on the machine

[image]

[image]

KalaNagHTD, 2023-01-27:
The Versions with 11kW up still have been using the old interface and controller. 
I´d try "ALTHERMA(LT_CA_CB_11-16KW)"

amasciotta, 2023-01-27:
> The Versions with 11kW up still have been using the old interface and controller. I´d try "ALTHERMA(LT_CA_CB_11-16KW)"

Should I use `#define PROTOCOL 'S'` or `#define PROTOCOL 'I'`?

KalaNagHTD, 2023-01-27:
I‘d use protocol „I“

amasciotta, 2023-01-28:
By following your suggestion and doing some experiments I noticed that there is no need to use the "S" protocol. The HP answers to the "I" protocol as well.

Moreover, not all the requested parameters get a response of `0x15 0xEA`. Mostly the regsitries with highest number (0x60, 0x61...) fail, but also some of the smallest ones.

Now I need to experiment a little bit more to find out the exact list of parameters that my HP supports. Any suggestion about this topic?

KalaNagHTD, 2023-01-28:
Experimenting is the best you can do imho. That‘s what I did (8kW version)

amasciotta, 2023-01-30:
The main problem at the moment is that I can't get the flow rate of the water.

If I uncomment the line
```
{0x62,9,105,2,-1,"Flow sensor (l/min)"}
```
the HP replies with `0x15 0xEA`.

That value is measured for sure by the machine, because I can read it through its display. But I can't get it through ESPAltherma... At this point I might give a try to DChecker, to see if I can get any more insight.

d3zit, 2023-05-07:
I think it's the same issue https://github.com/raomin/ESPAltherma/issues/59

No solution yet unfortunately