raomin/ESPAltherma #243: External Thermostat doesn't Update

Meatballs1, 2023-02-07:
The ext thermostat settings dont seem to update, despite other registers doing so?

They are both currently off and have been for 30 minutes or so. I've checked the unit itself and they both show as off in the Actuators menu.

However, ATTR updates are still showing them as:

```
  "Thermostat1 ON/OFF": "ON",
  "Thermostat2 ON/OFF": "ON",
```

Everything else seems to update fine. There doesn't appear to be any errors in the MQTT messages:

```
Querying register 0x10... 
Querying register 0x60... 
```

I'm using the Altherma LT 11-16kW hydrosplit header file:

```
{0x10,1,307,1,-1,"Thermostat1 ON/OFF"},
{0x60,2,303,1,-1,"Thermostat2 ON/OFF"},
```

They do sometimes correctly update to show as OFF or ON... But not often.

Meatballs1, 2023-02-07:
Maybe related to comments in https://github.com/raomin/ESPAltherma/issues/153

javimurcia, 2023-03-07:
I think i have encountered this in passign.

This value is only updated when you choose "external thermostat control" on your indoor unit.