raomin/ESPAltherma #245: What is the use case of `dataType` of the class `LabelDef`?

abutscher, 2023-02-07:
Hi,

What is the actual meaning and use case of the `dataType` member in the class `LabelDef` (https://github.com/raomin/ESPAltherma/blob/ebc7757dab8fa4b319b8bda0fd45d13b39724c3e/include/labeldef.h#L13)?
I can only see it used in the constructor to initialize its pointer counterpart. But then, it looks as if it's not actually used in the application source code. In the defining header files (the model specific ones), it seems to adopt either `1` or `-1` as value.
So, is `dataType` really used in the source code or am I just overlooking something?

BR

Tardymo, 2023-02-09:
Yes, it looks like dataType is unused.
And regarding InfluxDB, there is no need to write an application as Telegraf can parse JSON and write it to InfluxDB. That is exactly what I am doing right now. Just add this to telegraf.conf in INPUT PLUGINS section:

```
[[inputs.mqtt_consumer]]
  servers = ["tcp://mosquitto:1883"]
  topics = [
    "espaltherma/ATTR",
  ]
  data_format = "json"
```
My mqtt broker is running in a container next to telegraf, hence I can specify container name instead of IP address of the broker. 
By default string fields are filtered out by mqtt_consumer, and if you want some of them stored in InfluxDB, add them to configuration like this:

```
  json_string_fields = [
    "Thermostat ON/OFF",
  ]
```
You can further customize the consumer to for example filter out some of the JSON fields, rename them e.t.c.

raomin, 2023-02-09:
Yes you're right. This parameter is not used.
I'll remove it in a next commit.

abutscher, 2023-02-09:
Thanks for your answers. Really appreciate it.
@Tardymo even better if there is already an existing solution :+1: . Wasn't aware of that.
@raomin Just for my understanding of the source code. Is it then true to say that the actual type of each registry entry is then "encoded" in its converted ID?

raomin, 2023-02-09:
Each value is encoded somewhere in the registry bytes. See it has a `packed struct` in C. The converter identified by the `convID`, is then taking the bytes and extract it to the right value. This can be a boolean, temperature, pressure etc...
So yes you can have a look at the different converter to know if the value is string of int or float. But that's easier to do it by hand for the exact values you want

Tardymo, 2023-02-09:
@raomin All header files will have to be re-generated, since Daikin has released new version of Dhecker, which finally solves the problem of choosing the right file, since file names list compatible heat pump models. I can try to do this since I successfully generated one for my Altherma model. Can I also remove unused dataType field from the structure? I can create a pull request then. By the way, there is one small pull request from me which you have not yet merged.

Tardymo, 2023-02-13:
Just checked and here is what dataType actually means:

1 - temperature value
2 - pressure value
3 - temperature difference value
-1 - all other values

So probably dataType should remain in header files, as this value could allow to configure units for temperature and pressure in some future ESPAltherma update.

raomin, 2023-03-12:
Hi @Tardymo.
Sorry for the late reply. I've been busy with other projects recently... and I didn't even have time to have a look at the new definition files, but I love what you're saying! This sounds great.
If you can make a PR with this, that would be fantastic. 

I've merged your PR on negative values, sorry again for being late...