raomin/ESPAltherma #248: Dead m5stick

MJankeN, 2023-02-14:
Hi
I have 2 dead m5stick. Both brand new. The first one started reading low battery, after the CODE. Turned OFF and never came back to life again.
The second one, CODE uploaded. Turned it OFF on the BUTTON. Dead it is.
Any help would be nice. 
Got two m5stick, a brand new Altherma, a Raspberry Pi with mqtt setup. All ready to go. But this program seem to take the best of me and the hardware.
Thanks

raomin, 2023-02-14:
Unlikely dead. Describe a bit more what you did and what you see when you plug them in or upload code. 
Plug the in USB to your pc. If you hear a Windows sound, they're not dead. 
It takes 7 long seconds to start from the button.

MJankeN, 2023-02-14:
Windows sound when plugged in, yes. Pushed the ON button formore then 15 sec. Nothing.

raomin, 2023-02-14:
What happens when you upload the code?

MJankeN, 2023-02-14:
Failed to connect. Error Code 2.

MJankeN, 2023-02-14:
Managed to upload the code, and then the problem started.

MJankeN, 2023-02-14:
Uploaded again. Success it says, but it is pitch dark in the screen and not connecting to wifi. LED used to flash, but not now.

raomin, 2023-02-14:
You're using M5StickC, or M5StickC+?
Make sure to select the right profile on the blue bar as said in the Readme [here](https://github.com/raomin/ESPAltherma#step-1-uploading-the-firmware) Step 2.

MJankeN, 2023-02-15:
Good point with the right profile. Checking later today.

raomin, 2023-02-15:
Good! Glad to hear. Have fun!

closed 2023-02-15 by raomin

MJankeN, 2023-02-15:
But. There is always a but. It came through on the MQTT. There is something missing in the entities. It only shows that it is online?

[image]