raomin/ESPAltherma #265: EHSXB08P30EF shows R7T on display, but can't find in config files

maromme, 2023-03-14:
Hi,
I'm trying to get the espaltherma working. Well actually the first steps went really smooth, it's about the finetuning. I've got an EHSXB08P30EF. On the display I can find that piping diagram:
[image]
Does anyone know how to get the R7T temperature? I'm using the config file for ALTHERMA(HPSU6_ULTRA)
Thanks a lot,
Markus

Tardymo, 2023-03-14:
Try this header file. This is from the latest DChecker version. I will try to release these files for everyone this weekend.
[Altherma(ERGA E EHSH-X P30-50 E_EF series 04-08kW-ECH2O).zip](https://github.com/raomin/ESPAltherma/files/10973805/Altherma.ERGA.E.EHSH-X.P30-50.E_EF.series.04-08kW-ECH2O.zip)

maromme, 2023-03-15:
you made my day :-) Thatk's a lot!
Here is my result:
[image]

and for everyone usage, the background picture:
[image]

and the corresponding code:
`type: picture-elements
elements:
  - type: state-label
    prefix: 'R6T = '
    entity: sensor.espaltherma_r6t
    style:
      top: 85%
      left: 30%
      transform: translate(-50%, -50%) scale(0.5, 0.5)
      font-size: 20px
  - type: state-label
    prefix: 'R5T = '
    entity: sensor.espaltherma_r5t
    style:
      top: 72%
      left: 76%
      transform: translate(-50%, -50%) scale(0.5, 0.5)
      font-size: 20px
  - type: state-label
    prefix: 'R2T = '
    entity: sensor.espaltherma_r2t
    style:
      top: 74%
      left: 47%
      transform: translate(-50%, -50%) scale(0.5, 0.5)
      font-size: 20px
  - type: state-label
    prefix: 'R1T = '
    entity: sensor.espaltherma_r1t
    style:
      top: 52%
      left: 33%
      transform: translate(-50%, -50%) scale(0.5, 0.5)
      font-size: 20px
  - type: state-label
    prefix: 'R4T = '
    entity: sensor.espaltherma_r4t
    style:
      top: 42%
      left: 33%
      transform: translate(-50%, -50%) scale(0.5, 0.5)
      font-size: 20px
  - type: state-label
    prefix: 'R3T = '
    entity: sensor.espaltherma_r3t
    style:
      top: 14.2%
      left: 9%
      transform: translate(-50%, -50%) scale(0.5, 0.5)
      font-size: 20px
  - type: state-label
    prefix: 'P = '
    entity: sensor.espaltherma_waterpressure
    style:
      top: 18%
      left: 50%
      transform: translate(-50%, -50%) scale(0.5, 0.5)
      font-size: 20px
  - type: state-label
    prefix: 'V = '
    entity: sensor.espaltherma_flow
    style:
      top: 027%
      left: 64%
      transform: translate(-50%, -50%) scale(0.5, 0.5)
      font-size: 20px
  - type: state-label
    prefix: 'R7T = '
    entity: sensor.espaltherma_r7t
    style:
      top: 57%
      left: 75%
      transform: translate(-50%, -50%) scale(0.5, 0.5)
      font-size: 20px
image: local/Rohrleitungsplan.png
`

closed 2023-03-15 by maromme