raomin/ESPAltherma #267: SCOP - Calculation

dariuszsq, 2023-03-27:
Hi,

Since we have all information to calculate COP which is really cool but we cannot rely fully on that information. 

As a enduser i would like to see my SCOP in the panel. Based on the attributes we cannot calculate it (correct me if i am wrong)

Can we expose some more information to be able to calculate it ? I would like to get:
Consume energy - it can be counter
Produce energy - also can be a counter.

ap20132022, 2023-04-04:
you can calculate it. It's now 100% correct but it's possible. Take a look under discussion.

dariuszsq, 2023-04-04:
Correct if i am wrong but discussion is about COP - which i already have in my dashboard.
in order to get SCOP we need to have produced (and consumed by home) and consumed (from grid) energuy