raomin/ESPAltherma #272: Support for Warm Water Heatpump EKHHE-PCV3

hwinkel, 2023-04-15:
Are the warm water heat pumps EKHHE-PCV3 EKHHE-CV3 also supported ?

blablaaddi, 2023-05-04:
Trying to find esp support as well. thy are also known as Daikin-Altherma-M-HW.

hwinkel, 2023-05-04:
Was also trying to find someone owns this WP to have a look on connectors and PCB.

hwinkel, 2023-05-06:
At least I found the PCB Description which show a CN22 for WLAN Module and some other connectors, but they have different names as the usual ones.

[image]

KalaNagHTD, 2023-05-06:
Looking at the PCB layout, terminal identifiers and parameter setup logic of this tdhw heatpump it seems to be very unlikely that ESPAltherma will support it.

donwolle, 2023-11-05:
hi, did anybody test it with this heat pump?

is there another way to get temperature information of atherma m hw 260?

donwolle, 2023-11-10:
[image]

Should i try the Red Port?

[image]

roboticsalign, 2024-05-13:
> [image]
> 
> Should i try the Red Port?
> 
> [image]
> 
> 

Did you try it?

raomin, 2024-05-14:
Looking at the description, I would give cn21 a try

roboticsalign, 2024-05-14:
It's the blue one. In my case, it's unconnected.
 [image]

roboticsalign, 2024-05-15:
It's the blue one. In my case, it's unconnected.
 [image]

It's not working. I compiled the default Header file and connected the esp32 to CN22 and CN21. In both cases I get a timeout error.

Now I operate in boost mode (heat pump and coil) with two switches: one is switching the coil power supply, the other one the Daikin in total. This way I can control [off; hp; HP + coil], but I have no access to the internal sensors. 

Do you have any recommendations for improvements?

roboticsalign, 2025-01-09:
I'm quite happy with the additional switches and did not do more research on a remote control. But it would be great to control the Altherma directly with access to the measurements...

Hansfreddy, 2025-01-09:
@roboticsalign to be sure about the additional switches: You disconnected CN16 and put a relay in between I guess? 

Maybe you want to keep an eye on https://github.com/raomin/ESPAltherma/discussions/162
I don't know if I will achieve anything, but plan B is to read the display with ESP32-CAM (to get at least one temperature and current status) and manipulate the buttons with an ESP32 as well (this will require some tinkering with the capacitive buttons those can be tricky to influence).

roboticsalign, 2025-01-09:
> @roboticsalign to be sure about the additional switches: You disconnected CN16 and put a relay in between I guess?
>
> Maybe you want to keep an eye on [...]

Exactly, CN16.

The only bad thing is with this setup, that I need an external temperature sensor.

Thanks for pointing me to the other thread. I'll keep an eye on it.

My Altherma activates the screen saver after a short time, so I would expect much effort to work with an esp32 cam.

principat, 2025-08-16:
may be it helps but i found an interesting comment in another discussion:

Daikin does not produce this, it only sells it with their own trade mark. The producer is ferroli and the EKHHE is an EGEA TECH LT.

These could be cascaded with the CN21. 

https://www.ferroli.com/media/3540000410_03_EGEA_TECH_200%20LT%20-%20260%20LT%20-%20200%20LT-S%20-%20260%20LT-S.pdf

So it seams the need to activate the cascade mode to enable the CN21 Port and than it may be possible to direct connect to port CN21.

raomin, 2026-10-23:
Closing with the 2.0 release: the EKHHE is made by Ferroli and speaks its own protocol, not the Altherma one. Thanks for the pointers in the thread.

closed 2026-10-23 by raomin