raomin/ESPAltherma #275: possibility for ETSX16P50D ?

G-Mein, 2023-04-26:
Hi, we got the Altherma 3 H HT ECH2O but I can't find the connection on the PCB and also no config about our pump in espaltherma. Is it possible to work with it or not? We are really in a bad situation, cause the official gateway only works with an app, that's not working on current android phones and so...

stomko11, 2023-04-26:
Where have you been looking?

raomin, 2023-04-28:
It looks like it should have one. Can you paste a picture of the main board?

rayxxxxx, 2023-04-29:
I've got the Altherma 3 H HT ECH2O too and I installed ESPAltherma
[image]

KalaNagHTD, 2023-04-29:
[https://github.com/raomin/ESPAltherma/discussions/274](url)

hattmann, 2023-11-13:
> I've got the Altherma 3 H HT ECH2O too and I installed ESPAltherma and all works fine !

> 've got the Altherma 3 H HT ECH2O too and I installed ESPAltherma and all works fine

@rayxxxxx May i ask you which definition file you used during the setup?  I too have the Altherma 3 H HT Echo2O with outdoor unit epra16dw17 with indoor unit ETSXB16P50EF7 and i am unsure which def file to take.

rayxxxxx, 2023-11-13:
> > 've got the Altherma 3 H HT ECH2O too and I installed ESPAltherma and all works fine
> 
> @rayxxxxx May i ask you which definition file you used during the setup? I too have the Altherma 3 H HT Echo2O with outdoor unit epra16dw17 with indoor unit ETSXB16P50EF7 and i am unsure which def file to take.

I use ALTHERMA(HPSU6_ULTRA).h. 
The main data work without issues just two points don't work: pressure def (I tried all possible defs of the other def files but nothing) and Backup heater (BUH) . About BUH, @KalaNagHTD explained me: "The data on X10A are provided by the Daikin board A1P. The HPSU/ECH2O has a second controller board BM1 (originally developed by Rotex/Kromschroeder) where the BUH is connected to. I learned that this is the reason why BUH operation and step1/step2 messages from A1P are not always consistent."

hattmann, 2023-11-14:
Thank you, i will give this a try!