raomin/ESPAltherma #280: compiling for ESP-12F ESP8266 fails

maromme, 2023-05-09:
Hi,
I try compiling for an ESP8266 but it fails. I took a fresh download of the code and still... I'm using the env:nodemcuv2
```
.pio\libdeps\nodemcuv2\PubSubClient\src\PubSubClient.cpp: In member function 'boolean PubSubClient::publish_P(const char*, const uint8_t*, unsigned int, boolean)':
.pio\libdeps\nodemcuv2\PubSubClient\src\PubSubClient.cpp:523:16: warning: comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]
  523 |     return (rc == expectedLength);
      |             ~~~^~~~~~~~~~~~~~~~~
In file included from src\main.cpp:22:
include/converters.h: In member function 'void Converter::convert(LabelDef*, unsigned char*)':
include/converters.h:228:35: warning: too many arguments for format [-Wformat-extra-args]
  228 |             sprintf(def->asString,"{0:X}{1:X}", num, num2);
      |                                   ^~~~~~~~~~~~
src\main.cpp: In function 'void setup_wifi()':
src\main.cpp:145:8: error: 'class ESP8266WiFiClass' has no member named 'setScanMethod'
  145 |   WiFi.setScanMethod(WIFI_ALL_CHANNEL_SCAN);
      |        ^~~~~~~~~~~~~
src\main.cpp:145:22: error: 'WIFI_ALL_CHANNEL_SCAN' was not declared in this scope
  145 |   WiFi.setScanMethod(WIFI_ALL_CHANNEL_SCAN);
      |                      ^~~~~~~~~~~~~~~~~~~~~
src\main.cpp:146:8: error: 'class ESP8266WiFiClass' has no member named 'setSortMethod'
  146 |   WiFi.setSortMethod(WIFI_CONNECT_AP_BY_SIGNAL);
      |        ^~~~~~~~~~~~~
src\main.cpp:146:22: error: 'WIFI_CONNECT_AP_BY_SIGNAL' was not declared in this scope
  146 |   WiFi.setSortMethod(WIFI_CONNECT_AP_BY_SIGNAL);
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\nodemcuv2\src\main.cpp.o] Error 1
```
Does someone got a hint for me?
Thanks,
Markus

raomin, 2023-05-09:
Hi, you probably have an older version of esp library. Sorry I should have conditioned these lines.
Anyway, I reverted the fix that does work and removed these lines.
You can now get the latest version, it will work.

closed 2023-05-09 by raomin

DieterB61, 2023-10-07:
I'm still get similar warnings (but no Error) when compiling code using nodemcuv2 environment.

`.pio\libdeps\nodemcuv2\PubSubClient\src\PubSubClient.cpp: In member function 'boolean PubSubClient::publish_P(const char*, const uint8_t*, unsigned int, boolean)':
.pio\libdeps\nodemcuv2\PubSubClient\src\PubSubClient.cpp:523:16: warning: comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]
  523 |     return (rc == expectedLength);
      |             ~~~^~~~~~~~~~~~~~~~~`
...
`In file included from src\main.cpp:22:
include/converters.h: In member function 'void Converter::convert(LabelDef*, unsigned char*)':
include/converters.h:240:35: warning: too many arguments for format [-Wformat-extra-args]
  240 |             sprintf(def->asString,"{0:X}{1:X}", num, num2);
      |                                   ^~~~~~~~~~~~
src\main.cpp: In function 'void updateValues(char)':
src\main.cpp:53:10: warning: variable 'alpha' set but not used [-Wunused-but-set-variable]
   53 |     bool alpha = false;
      |          ^~~~~`

Could this be a reason for my problems described in the discussion:
https://github.com/raomin/ESPAltherma/discussions/326