raomin/ESPAltherma #286: Wiring the relay

MathieuLH44, 2023-05-30:
Hello,

I plug my ESP32 on my heat pump on the X10A.
I can control my relay, but I have a doubt to connect it on the X2M

I check the voltage on the pin 1, 2, 1a, 2a, 3 4, on my French heat pump (they are equivalent to 35, 34, 35a, 34a, 31, 30 when I look UK video).
1 : 230V
2 : 230V
1a : 230V
2a : 230V
3 : 230V
4 : 0V

When I look on the schematic wiring, I have to connect the 1 and the 4, but I scared to blow my PCB

Someone can confirm to me the connection please ?

raomin, 2023-05-30:
That makes sense. The relay will connect some power (1) to something that will sense the power (4).
Now juste copy paste a picture of your schematic if you want us to check...

MathieuLH44, 2023-05-30:
[image]
https://mail.google.com/mail/u/0?ui=2&ik=44e677b3ba&attid=0.1&permmsgid=msg-a:r4229765610345409761&th=1886e66464a84089&view=att&disp=safe&realattid=1886e6618deeaeccb41
This is the schematic and the connectors

MathieuLH44, 2023-06-09:
Someone can help me ?

raomin, 2023-06-09:
That makes complete sense. 1 and 4 on the relay. Eg 1 to the NO and 4 to COM.

MathieuLH44, 2023-06-14:
Thanks, I will do it soon

Edsol, 2025-01-16:
I have the same doubt on EHYKOMB33AA model, I connect the COM on pin 4 andh NO on the pin 1 but nothing changes. There are any configurations to set to use it?