raomin/ESPAltherma #297: Adding cooling function

ap20132022, 2023-06-19:
we have now our summer in Europe. I use the relays for turn on, off the heating function, SG1 & SG2. I miss the cooling function. 
Does anyone need it?

Can anyone help me to add this function? 
Thank you very much.

gjonesy84, 2023-06-20:
Hi,

Ive been looking into this in the UK. We use to get few really hot days a year. Now we have a few hot weeks a year.

I currently have a Daikin Altherma EBHQ006BBV3 6kW outdoor unit connected to a EKCB*H*008BCV3 indoor unit. The H in the part number is heating only.

Anyway as you know the outdoor unit can Heat and cool. However they do make a indoor unit EKCB*X*008BCV3 which enables the heating and cooling function. But I cannot find out what the difference is?

I would love to enable the cooling function. Whats the part number of your indoor unit?

Cheers

Gareth

theailer, 2023-06-29:
Hi
I've been trying to add control for a second relay to add the cooling function.
I think I have the control bit down but how to get the MQTT message to show both relays in home assistant eludes me still.

theailer, 2023-06-29:
A annoying solution would be to add a second esp32 to act as the relay controller with esp home or similar ;)

ap20132022, 2023-06-30:
> A annoying solution would be to add a second esp32 to act as the relay controller with esp home or similar ;)
My problem is the power supply. That's why I don't want to use the second one. 

> Hi I've been trying to add control for a second relay to add the cooling function. I think I have the control bit down but how to get the MQTT message to show both relays in home assistant eludes me still.
Maybe can @raomin or  @javimurcia help us?

javimurcia, 2023-06-30:
> I currently have a Daikin Altherma EBHQ006BBV3 6kW outdoor unit connected to a EKCB_H_008BCV3 indoor unit. The H in the part number is heating only.
>
> Anyway as you know the outdoor unit can Heat and [...]

For a heat pump to be capable of producing cold, it needs to be reversible. In the daikin altherma ecosystem, this part is handled by the interior unit as you have already guessed, exterior units can operate in both modes, so long they are paired with a reversible interior unit.

So, long story short, you need a reversible interior unit. I can say about your unit, but at least for my model, there is a kit you can purchase a kit (no idea on the price, or what includes, since my interior unit is reversibe) that can upgrade a heat-only interior unit, to a reversible one.

Once you have a reversible unit, at least in my unit, you can change the mode in 2 ways:
- Manually: you can choose if the heat pump is to produce heat or cold, via the interior unit control panel.
- Automatically: you can choose at what exterior temperature the heat pump should be in cooling mode, and at what temperature it should be in heating mode.

Also, the exterior thermostat can be configured in 2 ways:

- Single wire: When this is enabled, the thermostat output only signals demand, when it's enabled, the heat pump will turn on, and produce whatever is set to produce (heat or cold)
- Dual wire: the interior unit has 2 thermostat inputs. One input signals the demand for heating, and the other demand for cooling. Only when the active demand is consisten with the heat pump mode, the heat pump will turn on.

With all that said, ESPAltherma does not need any adaptation in order to be compatible with reversible heat pumps, just set it up to single wire control, and use EPSAltherma output to choose wether the heat pump is on or off. You will need to walk to the heat pump to change the mode, but at least where i live (Spain) you will only need to do that 2 times a year.

theailer, 2023-06-30:
Ah, I thought that dual wire was for a normal relay and single wire for the digital io pcb or whatever, thanks for that bit of information. It will be perfect for my setup.
My heat pump is a monoblock(air/water) unit with no indoor module except for the user interface.

I think we got hung up on this text from the instructions:
Step 3 (optional) - Controling your Daikin Altherma heat pump

ESPAltherma cannot change the configuration values of the heat pump (see [FAQ](https://github.com/raomin/ESPAltherma#faq)). However, ESPAltherma can control a relay on MQTT that can simulate an external On Off thermostat. Doing so allows to remotely turn on/off the heating function of your heat pump. A **second** relay can be used to trigger the cooling function.

ap20132022, 2023-06-30:
@javimurcia : I need an other relay for 
[image]
It's the external On Off thermostat
On the software from @raomin we connect it to the heating function. The cooling function is open. I can also turn off external thermostat, but it's better to control it over mqtt

gjonesy84, 2023-06-30:
> there is a kit you can purchase a kit (no idea on the price, or what includes, since my interior unit is reversibe) that can upgrade a heat-only interior unit, to a reversible one.

Yes this is what I want to purchase. I called Daikin and the EKCB_X_008BCV3 is now obsolete. Bummer :(

theailer, 2023-07-17:
> @javimurcia : Thank you very much for your reply. Where can I find "single wire control". On my HP with (indoor + outdoor) unit I can only turn on and turn off external thermostat.
> 
> I need an other relay for [image] It's the external On Off thermostat (on the right, number 5) On the software from @raomin we connect it to the heating function. The cooling function is open. I can also turn off external thermostat, but it's better to control it over mqtt

I connected the cooling function to the same relay but on the other output.
So now I can't shut it off just toggle between cooling or heating. But now I am also using the smart grid functions so I can stop/start the heating that way instead.