raomin/ESPAltherma #313: ERROR "Current measured by CT sensor of L1":"Conv 161 not avail."

TRIROG, 2023-08-17:
I Enabled the following lines in ALTHERMA(TOP-GRADE),
```
{0x63,14,161,1,-1,"Current measured by CT sensor of L1"},
{0x63,15,161,1,-1,"Current measured by CT sensor of L2"},
{0x63,16,161,1,-1,"Current measured by CT sensor of L3"},
```
however output in MQTT is
```
"Current measured by CT sensor of L1":"Conv 161 not avail.",
"Current measured by CT sensor of L2":"Conv 161 not avail.",
"Current measured by CT sensor of L3":"Conv 161 not avail."
```

monsieurhannes, 2023-10-01:
I got the same issue with the same heat pump.
There is also a (TOP-GRADE-LARGE_E) and a (TOP-GRADE-SMALL_E) in the settings and I wonder if that changes anything?
I don't know where the CT sensors should be, if that's a standard feature of a Altherma 3H MT, but maybe they just don't exist on this model.
However, I can't see any other current readings either.

Did you find a solution?

monsieurhannes, 2023-10-01:
Here is my JSON-string from `espaltherma/ATTR` (haven't cleaned it yet, I'm waiting to see what's actually important when the heating is ON as well...:
```
{
  "Fan Data Qty" : 1,
  "Operation Mode" : "Fan Only",
  "Defrost Operation" : "OFF",
  "Low noise control" : "OFF",
  "Error type" : "Normal",
  "Error Code" : " 0",
  "Comp. INV Current Drop" : "OFF",
  "Comp. INV Current Protection Retry Qty" : "Conv 311 not avail.",
  "Outdoor air temp.(R1T)" : 16.1,
  "Heat exchanger mid-temp." : 16.4,
  "INV primary current (A)" : 0,
  "INV secondary current (A)" : 0,
  "Compressor outlet temperature" : 0,
  "Fan 1 (step)" : 0,
  "Fan 2 (step)" : 0,
  "Outdoor heat exchanger temp." : 0,
  "Target Discharge Temp." : 0,
  "Target port temperature" : 0,
  "I/U operation mode" : "DHW",
  "LW setpoint (main)" : 32.1,
  "BSH" : "OFF",
  "BUH Step1" : "OFF",
  "BUH Step2" : "OFF",
  "Water pump operation" : "OFF",
  "Leaving water temp. before BUH (R1T)" : 19.5,
  "Leaving water temp. after BUH (R2T)" : 19.8,
  "Refrig. Temp. liquid side (R3T)" : 17.5,
  "Inlet water temp.(R4T)" : 19,
  "DHW tank temp. (R5T)" : 57.5,
  "Indoor ambient temp. (R1T)" : 20,
  "Ext. indoor ambient sensor (R6T)" : 0,
  "Circulation pump operation" : "OFF",
  "Flow sensor (l/min)" : 0,
  "Water pressure" : 1,
  "Refrigerant pressure sensor" : -6.9,
  "Current measured by CT sensor of L1" : "Conv 161 not avail.",
  "Current measured by CT sensor of L2" : "Conv 161 not avail.",
  "Current measured by CT sensor of L3" : "Conv 161 not avail.",
  "Hybrid Heating Target Temp." : 0,
  "Boiler Heating Target Temp." : 0,
  "Mixed water temp." : 0,
  "2nd Domestic hot water temperature" : 57.4,
  "Target delta T heating" : 0,
  "Target delta T cooling" : 5,
  "WifiRSSI" : "-83dBm",
  "FreeMem" : "192824"
}
```

raomin, 2023-10-02:
I'll add the conv 161 soon.

raomin, 2023-10-02:
Done.

closed 2023-10-02 by raomin

TRIROG, 2023-10-06:
anybody has any luck with these? On my A 3 H MT i don't see any changes reported - just 0 for all L1, L2 and L3.

abmantis, 2024-03-11:
@TRIROG  found anything? I also got 0's.

putteg, 2024-12-03:
Still no luck with this?