raomin/ESPAltherma #315: "Time out! Check connection" advice

zeman, 2023-08-27:
I've seen that a time out error often means a bad cable. I've tried two different brand new jumper cables. I've also tried the older Rotex S protocol as someone in the installation thread with the same looking model used that. M5StickC+ seems to be happy running on the 5v so at least that connection is working. Any other advice on what to try next?

[image]

domoneyleon, 2023-09-06:
I've used the connectors you have above on one system and found that it worked but the connection constantly dropped.  A different connection method I found improved the connection but still dropped occasionally.

[image]

[image]

Once I insert the "Pins" into the female connectors I use a pliers to crimp the female connector around the pins to ensure a good contact between the two metal parts.  This then goes into the M5 Stich ports and the white plastic JST connector goes into the x10A port on the Heat pump.

Hope this helps

domoneyleon, 2023-09-06:
This is the item on amazon
[image]

[image]