raomin/ESPAltherma #316: Eco mode property not set

habitoti, 2023-08-28:
I have a Daikin Altherma 8 MT F, indoor model ETVH12S18EJ9W. I use the _ALTHERMA(BIZONE_CB_04-08KW).h_ setup.

When comfort mode is selected, the _Storage comfort_ property goes properly to ON. When mode switches to Eco, however, _Storage eco_ property stays OFF. 

```
{0x62,2,306,1,-1,"Storage eco ON/OFF"}
{0x62,2,305,1,-1,"Storage comfort ON/OFF"}
```

Any clue where this could be rooted?

domoneyleon, 2023-09-05:
The only thing I can think of is the combination of the setup file and your heat pump?  I'd say have a look in other setup files for the storage eco address that isn't 306 as above and maybe edit this new address into your current code?  Not sure tbh but that's the first thing that springs to my mind.

habitoti, 2023-09-05:
It's 306 for all configurations. Also strange that it works for _comfort_, but doesn't for _eco_...