raomin/ESPAltherma #322: Altherma 3 R ECH2O

danilo-quadrelli, 2023-09-16:
Ho installato un ESPAltherma su Daikin 3 R ECH2O e tutto funziona. Non avendo il file iclude dedicato per il mio modello, ho usato ALTHERMA(HPSU6_ULTRA).h ed ho provato anche il DEFAULT.h della dir include/def/Italian. Ho molti dubbi sui valori che mi vengono restituiti.

Vi chiedo se qualcuno è riuscito a decodificare i dati per il mio modello di pompa di calore e la possibilità di avere un file include per Daikin 3R ECH2O.
Software N. B1/U1: 427.001.AF
Software N. controller: 425.002.D

Faccio i complimenti per lo splendito lavoro eseguito completo in ogni sua parte. 
Si potrebbe anche pensare di inviare qualche comando via seriale alla pompa di calore come il passaggio in raffrescamento, riscaldamento, stan by......

Grazie Danilo
[image]
[image]
[image]

tsrandrei, 2023-10-17:
@danilo-quadrelli sarebbe piu utile descrivere il problema in inglese, per essere comprensibile a piu utenti, magari qualcuno potrebbe aiutare.

mmariotti, 2023-12-30:
Ciao Danilo,

Io ho installato l'ESP in questo momento, e qualche dato sensato lo ricevo, usando il file "ALTHERMA(LT_DA_04-08KW).h".
Io ho una Altherma 3 R ECH2O (detta "Compact") con EU: ERGA08EVA1 e IU: EHSXBDP50DA1, mi è sembrato il file più adatto.

Di seguito il risultato della prima prova:

```
{
    "2way valve(On:Heat_Off:Cool)": "OFF", 			// WRONG!
    "3way valve(On:DHW_Off:Space)": "OFF",			// WRONG!
    "Brine Flow Switch": "ON",						// ???
    "Defrost Operation": "OFF",						// maybe
    "DHW setpoint": 54,								// CORRECT
    "DHW tank temp. (R5T)": 54.1,					// CORRECT
    "Ext. indoor ambient sensor (R6T)": 13,			// CORRECT
    "Flow sensor (l/min)": 17.5,					// CORRECT
    "FreeMem": "226644",							// n/a
    "I/U operation mode": "Heating",				// CORRECT	
    "Indoor ambient temp. (R1T)": 0.4,				// TBV
    "Inlet water temp.(R4T)": 27,					// CORRECT	
    "INV primary current (A)": 1.8,					// TBV
    "Leaving water temp. after BUH (R2T)": 29.5,	// CORRECT	
    "Leaving water temp. before BUH (R1T)": 29.5,	// CORRECT	
    "Liquid pipe temp.(R6T)": 0,					// WRONG!	
    "LW setpoint (main)": 28,						// CORRECT	
    "Operation Mode": "Heating",					// CORRECT	
    "Outdoor air temp.(R1T)": 14.5,					// TBV	
    "Pressure sensor": 0,							// WRONG!
    "Pressure sensor(T)": "Conv 405 not avail.",	// WRONG!
    "Refrig. Temp. liquid side (R3T)": 28.7,		// TBV
    "Solar input": "OFF",							// maybe
    "Solar pump operation": "OFF",					// maybe
    "Voltage (N-phase) (V)": 223,					// TBV
    "Water pressure": 0,							// WRONG!
    "Water pump operation": "ON",					// CORRECT	
    "WifiRSSI": "-65dBm"							// n/a
}
```

Appena ho tempo, ho intenzione di verificare il file della issue #350 

E, eventualmente, fare un minimo di reverse engineering per ricavare tutte le altre mappature possibili. Ad esempio, manca completamente la mappatura delle statistiche (energia), che probabilmente c'è, da qualche parte

danilo-quadrelli, 2023-12-31:
Good Morning Sir,
thanks for considering the problem, I really appreciate it.
I'm waiting for news
Have a good day, and many wishes for the New Year

simonebortolin, 2024-01-01:
@mmariotti 
> `"Pressure sensor(T)": "Conv 405 not avail.",	// WRONG!`

this is wrong becasue you need to apply this patch https://github.com/raomin/ESPAltherma/pull/325

> `"Liquid pipe temp.(R6T)": 0,					// WRONG!`

R6T is the Outdoor air temp, and the Liquid pipe temp is other thing

> `"Water pressure": 0,							// WRONG!`

`{0x62,11,105,1,1,"Water pressure"},`

This row for my 3R ECH2O (generation E, last, Daikin Eye) work.

mmariotti, 2024-01-01:
@simonebortolin 

> > `"Pressure sensor(T)": "Conv 405 not avail.",	// WRONG!`
> 
> this is wrong becasue you need to apply this patch #325

thanks, it is indeed a bug.  That pull should be merged.

> > `"Liquid pipe temp.(R6T)": 0,					// WRONG!`
> 
> R6T is the Outdoor air temp, and the Liquid pipe temp is other thing

My assumption is based on the fact that the value is always 0.
I have yet to check the meaning of 3-letter codes, good to know that there's also some semantic mistake.

> > `"Water pressure": 0,							// WRONG!`
> 
> `{0x62,11,105,1,1,"Water pressure"},`
> 
> This row for my 3R ECH2O (generation E, last, Daikin Eye) work.

In the file I used there's the mapping

`{0x62,11,105,1,2,"Water pressure"}`

Registry, offset, converter and data-size are the same. But data-type is different: as stated in the docs:
> Datatype: 1: temperature in (C), 2: pressure in (kgcm2), -1: everything else

I don't know yet what's the purpose of 'data-type' and how it is used (since there's a converter), but I'll try to chenage it to 1 and see what's the result.

Thank you for all the infos!

simonebortolin, 2024-01-01:
> thanks, it is indeed a bug. That pull should be merged.

I agree

> My assumption is based on the fact that the value is always 0.
> I have yet to check the meaning of 3-letter codes, good to know that there's also some semantic mistake.

`{0x61,12,105,2,1,"Indoor ambient temp. (R1T)"},`

This temp is provider by madoka, but dosen't have any 3-letter codeds into the HPU

> In the file I used there's the mapping
> 
> `{0x62,11,105,1,2,"Water pressure"}`
> Registry, offset, converter and data-size are the same. But data-type is different: as stated in the docs:
>
> > Datatype: 1: temperature in (C), 2: pressure in (kgcm2), -1: everything else

But.... this work with 2:
```
{0x61,2,105,2,1,"Leaving water temp. before BUH (R1T)"},
{0x61,4,105,2,1,"Leaving water temp. after BUH (R2T)"},
{0x61,6,105,2,1,"Refrig. Temp. liquid side (R3T)"},
{0x61,8,105,2,1,"Inlet water temp.(R4T)"},
{0x61,10,105,2,1,"DHW tank temp. (R5T)"},
{0x61,12,105,2,1,"Indoor ambient temp. (R1T)"},
{0x61,14,105,2,1,"Ext. indoor ambient sensor (R6T)"},
```

For this value, this work see https://github.com/raomin/ESPAltherma/issues/350#issuecomment-1873278084 (the photo and the screen have ten minut delay)

mmariotti, 2024-01-05:
I dumped raw data for about 2 days. "Water pressure" is definitely **not** at `0x62.11` as that byte is always `0x00`.
I'll try to check other unmapped registers (if exist) in the next days.

Meanwhile, the dumps are attached.
And this is the definition list I currently use:

```
{0x10, 0, 217, 1, -1, 		"Operation Mode"},
{0x10, 1, 300, 1, -1, 		"Low Noise"},
{0x10, 1, 301, 1, -1, 		"Demand Signal"},
{0x10, 1, 302, 1, -1, 		"Pressure Equalizing Operation"},
{0x10, 1, 303, 1, -1, 		"Oil Return Operation"},
{0x10, 1, 304, 1, -1, 		"Defrost Operation"},
{0x10, 1, 305, 1, -1, 		"Startup Control"},
{0x10, 1, 306, 1, -1, 		"Restart standby"},
{0x10, 1, 307, 1, -1, 		"Thermostat"},
{0x10, 4, 203, 1, -1, 		"Error type"},
{0x10, 5, 204, 1, -1, 		"Error Code"},
{0x10, 6, 114, 2, 1, 		"Target Evap. Temp."},
{0x20, 0, 105, 2, 1, 		"Outdoor Air Temp."},
{0x20, 4, 105, 2, 1, 		"Discharge Pipe Temp."},
{0x20, 8, 105, 2, 1, 		"Heat Exchanger Mid-Temp."},
{0x20, 16, 106, 2, 2, 		"Pressure (T)"},

{0x21, 0, 105, 2, -1, 		"Primary Current (A)"},
{0x21, 2, 105, 2, -1, 		"Secondary current (A)"},
{0x21, 4, 101, 2, -1, 		"Voltage (V)"},

{0x60, 2, 303, 1, -1, 		"Ext. Thermostat"},
{0x60, 2, 315, 1, -1, 		"I/U operation mode"},
{0x60, 3, 204, 1, -1, 		"Error Code"},
{0x60, 4, 152, 1, -1, 		"Error detailed code"},
{0x60, 5, 203, 1, -1, 		"Error type"},
{0x60, 7, 105, 2, 1, 		"DHW setpoint"},
{0x60, 9, 105, 2, 1, 		"LW setpoint"},
{0x60, 11, 303, 1, -1, 		"Solar input"},
{0x60, 11, 307, 1, -1, 		"Flow Switch"},
{0x60, 12, 300, 1, -1, 		"Solar pump operation"},
{0x60, 12, 301, 1, -1, 		"Water pump operation"},
{0x60, 12, 306, 1, -1, 		"3way Valve (On:DHW|Off:Space)"},
{0x60, 12, 307, 1, -1, 		"3way Valve (On:Heat|Off:Cool)"},

{0x61, 2, 105, 2, 1, 		"Leaving water temp."},
{0x61, 4, 105, 2, 1, 		"Leaving water temp. after BUH"},
{0x61, 6, 105, 2, 1, 		"Refrigerant temp."},
{0x61, 8, 105, 2, 1, 		"Returning water temp."},
{0x61, 10, 105, 2, 1, 		"DHW tank temp."},
{0x61, 12, 105, 2, 1, 		"Indoor temp."},
{0x61, 14, 105, 2, 1, 		"Outdoor aux. temp."},

{0x62, 3, 105, 2, 1, 		"LW setpoint (add)"},
{0x62, 5, 105, 2, 1, 		"RT setpoint"},
{0x62, 9, 105, 2, -1, 		"Flow sensor (l/min)"},
{0x62, 12, 152, 1, -1, 		"Water pump signal (0:max-100:stop)"},

{0x64, 7, 105, 2, 1, 		"LW Target Temp."},
{0x64, 12, 105, 2, 1, 		"2nd DHW temp."},
{0x64, 14, 152, 1, -1, 		"Target delta T heating"},
{0x64, 15, 152, 1, -1, 		"Target delta T cooling"},
```

[altherma_0x10.txt](https://github.com/raomin/ESPAltherma/files/13839532/altherma_0x10.txt)
[altherma_0x11.txt](https://github.com/raomin/ESPAltherma/files/13839533/altherma_0x11.txt)
[altherma_0x20.txt](https://github.com/raomin/ESPAltherma/files/13839534/altherma_0x20.txt)
[altherma_0x21.txt](https://github.com/raomin/ESPAltherma/files/13839535/altherma_0x21.txt)
[altherma_0x30.txt](https://github.com/raomin/ESPAltherma/files/13839536/altherma_0x30.txt)
[altherma_0x60.txt](https://github.com/raomin/ESPAltherma/files/13839537/altherma_0x60.txt)
[altherma_0x61.txt](https://github.com/raomin/ESPAltherma/files/13839538/altherma_0x61.txt)
[altherma_0x62.txt](https://github.com/raomin/ESPAltherma/files/13839541/altherma_0x62.txt)
[altherma_0x63.txt](https://github.com/raomin/ESPAltherma/files/13839544/altherma_0x63.txt)
[altherma_0x64.txt](https://github.com/raomin/ESPAltherma/files/13839547/altherma_0x64.txt)
[altherma_registry_values_summary.txt](https://github.com/raomin/ESPAltherma/files/13839550/altherma_registry_values_summary.txt)

simonebortolin, 2024-01-05:
```
{0x20, 0, 105, 2, 1, 		"Outdoor Air Temp."},
```

This not work in ECH2O gen E
```
{0x21, 4, 101, 2, -1, 		"Voltage (V)"},
```
This for me has a different value from reality, most probably missing a multiplication value like 0.5
```
{0x30,0,152,1,-1,"INV frequency (rps)"},
{0x30,1,211,1,-1,"Fan 1 (step)"},
{0x30,2,211,1,-1,"Fan 2 (step)"}, // only if is a dual fan E.U.
{0x60,2,302,1,-1,"Freeze Protection"},
{0x60,2,301,1,-1,"Silent Mode"},
```
Can you also try this?

mmariotti, 2024-01-05:
Mine is EHSXB08P50**D**, so gen D.
Registry offsets and/or values may be different from gen D to gen E.

As you can see from the dump data summary for registry 0x30:

```
work\altherma_0x30.txt
0 => [0x10, 0x0e, 0x16, 0x11, 0x0f, 0x14, 0x15, 0x12, 0x00, 0x23, 0x24, 0x1c, 0x1f, 0x1b, 0x18, 0x20, 0x17, 0x1a, 0x0c, 0x13, 0x0d, 0x1d, 0x2c, 0x34, 0x31, 0x21, 0x25, 0x08, 0x26, 0x36, 0x30, 0x2b, 0x27, 0x2d, 0x38, 0x1e, 0x0b, 0x0a, 0x2a, 0x2e, 0x2f, 0x28, 0x3a, 0x39, 0x22, 0x19]
1 => [0x43, 0x44, 0x3c, 0x00, 0x17, 0x30, 0x49, 0x0b, 0x45, 0x3f, 0x09]
2 => [0x00]
```

applying the specified converters, data becomes:

```
"INV frequency (rps)" 	0 => [16, 14, 22, 17, 15, 20, 21, 18, 0, 35, 36, 28, 31, 27, 24, 32, 23, 26, 12, 19, 13, 29, 44, 52, 49, 33, 37, 8, 38, 54, 48, 43, 39, 45, 56, 30, 11, 10, 42, 46, 47, 40, 58, 57, 34, 25]
"Fan 1 (step)" 			1 => [67, 68, 60, 0, 23, 48, 73, 11, 69, 63, 9]
"Fan 2 (step)"			2 => [0]
```

I can't tell what this data represents...  What's INV freq? Is the magnitude of values compatible with rps? I don't know.
The very same applies to Fan. Is it speed, rps, rpm? or maybe modulation %? I don't know either.
Since these data are not present on the GUI of the HP, I can't verify/validate (not even reference) them.

As of `0x60`, the dump contains values:

`[0x18, 0x00, 0x48, 0x40]`

using flag conversion 300, becomes:

`0001 0100
0000 0000
0010 0100
0010 0000`

choosing flags 301 and 302 (from left to right: the 2nd and the 3rd): 
- Silent Mode (301) is always FALSE: maybe it's correct, I have not changed this setting
- Freeze Protection (302) is half-and-half: probably wrong, I have not changed this setting (if it refers to the setting) and here the outdoor temperature has not been cold enough to trigger freeze protection.

So, in conclusion:

- I cannot tell if these mappings for `0x30` are correct as there's not comparable evidence on the GUI
- The mappings for `0x60` are probably wrong for my HP

raomin, 2024-01-06:
I updated the definition files. There are some for the ECH2O series. You should have a look.

mmariotti, 2024-01-07:
I found a way to connect D-checker with ESPAltherma using a simple MITM proxy that subscribes to MQTT logs and caches raw data, while sending it back, on request, to D-checker.

Using this method I can verify, and somehow "decrypt" (fuzzing-n-spoofing eventually) the definitions for, virtually, every model.

I checked that, for my model, the address of "water pressure" is correct (0x61,12) but my HP does provide always a 0.

When I'll have some spare time I'll check more values and share the MITM source.

simonebortolin, 2024-01-07:
Can you share the D-checker database?

mmariotti, 2024-01-07:
I can't access D-checker DB directly, but only indirectly. I can emulate an HP so that D-checker thinks it's talking to a real HP.

You can see all the definitions D-checker loads for the selected HP model (ERGA_E_EHSH-X_P30-50_E_EF_series_04-08kW-ECH2O). I've chosen gen **E**, so you can see all the definitions for your HP model.

Nevertheless, you can't see register and offeset for each definition directly: you have to make the spoofer send specifically crafted data to D-checker and see how the value is interpreted in the GUI.

I attach some screen, at least you can see the definitions.

@raomin 
In the last screen you can see what D-checker is asking. The last questions are made up of 5 bytes instead of 4:

`Q: 04:40:70:00:4B`

Do you know how to interpret them? Are there 2-bytes registers?

[image]
[image]
[image]
[image]
[image]
[image]
[image]

simonebortolin, 2024-01-07:
> I updated the definition files. There are some for the ECH2O series. You should have a look.

I am trying the new file and find the error in water pressure to be conv 105 and not 405. Or is conv 405 wrong?

bzq-pl, 2024-01-08:
@mmariotti,
Could you be so kind and share details how do you emulate Heat Pump to make d-checker running without real HP connected?

mmariotti, 2024-01-08:
Sure, I'll do it as soon as I can.
To create the spoofer quickly I had to implement it using my company proprietary framework and I need to extract the spoofer to a standalone application (I cannot share company code).
I plan to do it in a couple of days

bzq-pl, 2024-01-08:
Grazie mille :-)

bzq-pl, 2024-01-09:
thank you! - will review this evening.