raomin/ESPAltherma #328: Support for Altherma 3 EDLA

arc12, 2023-10-10:
Is this unit supported? https://www.daikin.co.uk/en_gb/products/product.html/edla04-08e3v3.html
I consulted setup.h but none of the options matched.

Thanks in advance...

IamMattM, 2023-10-13:
I'll be interested to verify this as well as I expect to have this same Altherma 3 EDLA monoblock model installed in the not too distant future by Octupus Energy UK.

Alibloke, 2023-10-18:
It appears to be, just not listed in the setup.h file explicitly:
https://github.com/raomin/ESPAltherma/blob/main/include/def/ALTHERMA(EBLA-EDLA%20%20EWAA-EWYA%20D%20SERIES%209-16KW).h

pokery, 2023-11-09:
I got it working with the EDLA09DA3V3 in the UK which is essentially the same unit but a different size I think.

The only thing that threw me was that the voltage value ("Voltage (N-phase) (V)") doesn't seem to be in any of the locations mentioned in various issues on here. If you're wanting to calculate the COP, you can hard code the UK voltage or use another HA sensor (ours is using the live voltage from our Powerwall).

greem, 2023-12-28:
I've just created a PR for my changes to the def file for an EDLA09DA3V3 installed by Octopus Energy.

https://github.com/raomin/ESPAltherma/pull/372

As stated in the PR, the 'Voltage (V)' value needs work but the important bits like flow rates, inlet/outlet temps etc seem sane and match the basic info provided by the indoor unit/controller and the bits reported in the Onecta app or HomeAssistant integration.

raomin, 2024-01-05:
@simonebortolin suggested that the voltage was actually double.
Can you check if using this as definition for the voltage gives accurate result?
`{0x21,4,111,2,-1,"Voltage (V)"},`

Converter 111 would make `*0.5` for the value.

simonebortolin, 2024-01-05:
You are using the wrong definition of Voltage (V) (0x21,4) that has already been established. Since the register is the same of the ampere. 

Use the other one (0x21,6) and divide by 0.5 or 0.7.

simonebortolin, 2024-01-05:
> @simonebortolin suggested that the voltage was actually double. Can you check if using this as definition for the voltage gives accurate result? `{0x21,4,111,2,-1,"Voltage (V)"},`
> 
> Converter 111 would make `*0.5` for the value.

But i'm not sure... I need some test...

simonebortolin, 2024-01-07:
@raomin 

with the new files of the new version of d-checker 0x21,4,105,2,1,"INV fin temp."

swalsh11, 2024-02-29:
I am logging the output for the 3way valve, to try and determine when DHW is called,  however, the signal never changes, IS anyone else seeing this? or is there another way to see DHW?

greem, 2024-02-29:
> I am logging the output for the 3way valve, to try and determine when DHW is called, however, the signal never changes, IS anyone else seeing this? or is there another way to see DHW?

I've just created a template binary sensor following the attribute "3way valve(On:DHW Off:Space)", which will next change state (in theory!) tomorrow morning when the schedule runs. Unless the kids get home and use all the water showering after school :)

swalsh11, 2024-03-02:
Looks like I Made a mess of my template.yaml, I have it working now.

abmantis, 2024-03-13:
@simonebortolin  why you mention 0.7 for INV fin temp?
Also, isn't INV fin temp the temperature of the invertor board fin?

simonebortolin, 2024-03-13:
> @simonebortolin why you mention 0.7 for INV fin temp? Also, isn't INV fin temp the temperature of the inverter board fin?

Because it was thought to be the tension.

abmantis, 2024-03-13:
> > @simonebortolin why you mention 0.7 for INV fin temp? Also, isn't INV fin temp the temperature of the inverter board fin?
> 
> Because it was thought to be the tension.

But why 0.7?

simonebortolin, 2024-03-13:
> > > @simonebortolin why you mention 0.7 for INV fin temp? Also, isn't INV fin temp the temperature of the inverter board fin?
> > 
> > 
> > Because it was thought to be the tension.
> 
> But why 0.7?

Please read old message. I wrote that I thought that value if it was really the voltage (which it isn't!!!!) was to be reduced by 0.5 or 0.7 (sqrt(2)/2, which if you know how AC voltage works you know why). 

Now there is no point in applying this value since we finally know that this figure is a temperature and not a voltage.

Now what temperature it represents I do not care, it is an internal temperature of the inverter, I do not care more than that.

abmantis, 2024-03-13:
> Please read old message. I wrote that I thought that value if it was really the voltage (which it isn't!!!!) was to be reduced by 0.5 or 0.7 (sqrt(2)/2, which if you know how AC voltage works you [...]

I already read the old comments, that is why I was asking "why 0.7 (sqrt(2)/2)?". I don't "know AC voltage", otherwise I wouldn't be asking. I was just curious about where that value comes from. 

If that value is not voltage and EDLA is supported (that I can confirm), this issue can be closed I guess.

acabadora, 2024-03-17:
is EBLA09DV3 same model with the one of discussion in this topic? and does it share compatibility?

greem, 2024-04-15:
> Looks like I Made a mess of my template.yaml, I have it working now.

Can you tell us what you did? If I refer to the state_attr for '3way valve(On:DHW Off:Space)', I always get null as a result.

swalsh11, 2024-04-15:
Try this

#template.yaml
  - sensor: # this line is needed only once
      - name: "3way valve(On:DHW Off:Space)"
        unique_id: "3way_valve_mode"
        icon: mdi:valve
        state: >
          {% if is_state_attr('sensor.althermasensors', '3way
valve(On:DHW_Off:Space)', 'ON') %}
            1
          {% else %}
            0
          {% endif %}

greem, 2024-04-16:
Dammit! The underscore. Name has "3way valve(On:DHW Off:Space)" but the sensor has "3way valve(On:DHW_Off:Space)".

I feel rather daft now! Thanks for opening my eyes to a rather obvious typo.