raomin/ESPAltherma #338: Incorrect DHW tank temp

luuks123, 2023-10-26:
I get an incorrect notation in Home Assistant on my DHW tank temperature, what could be the cause of this?

`Water pump operation: 'OFF'
Indoor Option Code: 11
Outlet Water Heat Exch. Temp. (R1T): 28.5
Outlet Water BUH Temp. (R2T): 27.4
Refrig. Temp. liquid side (R3T): 23.7
Inlet water temp.(R4T): 22.9
DHW tank temp. (R5T): -40.7
RT Temp.: 20.8
Powerful DHW Operation. ON/OFF: 'OFF'
Space heating Operation ON/OFF: 'ON'
Emergency (indoor) active/not active: 'OFF'
LW setpoint (add): 38.5
RT setpoint: 19.5
DHW setpoint: 65
LW setpoint (main): 25
Water flow switch: 'ON'`

TungstenE2, 2026-09-07:
mine on Rotex is correct

[image]

raomin, 2026-10-23:
@luuks123 It's been a while! ESPAltherma 2 handles it: when the tank sensor answers -40.7 (no sensor), the tank simply disappears from the drawing and the value isn't selected by default. Closing with the 2.0 release.

closed 2026-10-23 by raomin