raomin/ESPAltherma #339: How to get registers for COP?

membersound, 2023-10-27:
What is the register for "consumed energy kWh" and the "energy delivered"?
The values are visible in the touch menu of my altherma, so they should be readable.

I want to calculate the COP, but I would need those values for it. But where are they? I looked into all templates, but none of them seem to provide this information?

jeroenvdheuvel, 2023-10-30:
I would love this data too, I'm not sure how to and if we can get it. For the time being there's a way to calculate the COP: https://github.com/raomin/ESPAltherma#calculating-cop

skrgahr, 2023-12-14:
That would be great for me as well. That way i can easily calculate and track SCOP, that i basically care more about than COP.

DraycD, 2025-02-28:
This Values would be awesome, how can we help to get this data ?

raomin, 2026-10-23:
It's been a while! The energy counters of the interface are not on the port. ESPAltherma 2's doc has the COP from the flow and the water temperatures (doc/home-assistant.md), and an on-board estimate from the inverter current is on the list. Closing.

closed 2026-10-23 by raomin