raomin/ESPAltherma #340: I don' see history for my attributes

gattigianluca, 2023-10-29:
Hi all,
I had to re-install homeassistant but, my bad, I have not done backup of configuration.yaml.
I have configured it again but now i am not able to see anymore the history of the attributes I have entered in the configuration.yaml.
What I am doing wrong? I'm getting mad 👎 
Attached the configuration.yaml and the history I see in HA
Thanks for the ones willing to help me!

[image]
[configuration.yaml.txt](https://github.com/raomin/ESPAltherma/files/13197845/configuration.yaml.txt)

jeroenvdheuvel, 2023-10-30:
Did you enable the [Home assistant recorder](https://www.home-assistant.io/integrations/recorder/)? You can configure the number of days to keep via the argument: `purge_keep_days`.

gattigianluca, 2023-10-30:
Yes it is enable by the default_config:  that is at the beginnin of the configuration.yaml

gattigianluca, 2023-11-01:
Solved, I was calling wrongly the entity :-

closed 2023-11-01 by gattigianluca