raomin/ESPAltherma #347: Daikin Altherma 3 EDLA08E2V3

youmustcomply, 2023-11-07:
I had this heat pump system installation installed in the summer by Octopus Energy. However I am finding one Onecta application somewhat lacking in information. I have intergrated this into HomeAssistant which gives slightly more information but still lacks.

I ordered a DIYMORE ESP32 and installed ESPAltherma onto it succesfully. Connected this up to the X10A port on the main board and restarted the heat pump. 

However; the main controller showed a U8-07 P1P2 Communication Error. I checked all the connections and restarted again and this time the unit just sat saying synchorising with Internal unit and stayed at 0%.

Any ideas?

youmustcomply, 2023-11-07:
Okay, after checking all the connections I realised I had connected what should of been the Ground on the ESP to a 3.3V input. No idea how i did this. I will re-test tomorrow.

Alibloke, 2023-11-28:
@youmustcomply can you tell me which definition file you used with the unit?  Was it ALTHERMA(LT_DA_04-08KW).h?

youmustcomply, 2023-11-29:
@Alibloke I used the default one and then just picked off the measurements I wanted.

Alibloke, 2023-11-29:
@youmustcomply Thanks!  Slightly off topic, but I'm about to get the 8kW unit installed from Octopus in just under three weeks.  How is it for you?

youmustcomply, 2023-11-29:
@Alibloke Yes that is what I had installed back in June. The power use has just started increasing alot with the low temps. Worst I have seen is 20kW in a day for heat. House is running pure weather compensation so I am just using a weather curve for the flow temp. I knock this back 3 points from 7pm till 3am.

neildsb, 2023-12-12:
Hi @youmustcomply sorry to be a pain, I also have an 8kW unit from Octopus and was disappointed with the MMI/app and lack of COP calculation, just connected up the ESPAltherma with an M5StickCPlus

I have problems with exposing additional attributes, do I have the data pins the wrong way around (the power is good 4.85V)? 

[image]

youmustcomply, 2023-12-12:
> Hi @youmustcomply sorry to be a pain, I also have an 8kW unit from Octopus and was disappointed with the MMI/app and lack of COP calculation, just connected up the ESPAltherma with an M5StickCPlus [...]

Have you un-commented the attributes you are wanting to see within the definition file you are using?

neildsb, 2023-12-12:
TBH it's my 1st time using PlatformIO (ESPHome is my go to in home assistant), yes I think have uncommented some attributes 

[image]

youmustcomply, 2023-12-12:
Thats good, there are a lot of duplicates and a lot don't work. I am at work tonight but I should be able to have a look tomorrow and see which ones I have used and let you know.

neildsb, 2023-12-12:
> Thats good, there are a lot of duplicates and a lot don't work. I am at work tonight but I should be able to have a look tomorrow and see which ones I have used and let you know.

really sorry, got my TX and RX pins the wrong way around; if you could share the useful attributes for the 8kW unit that would be fab, have a great evening

[image]

youmustcomply, 2023-12-14:
These are the ones i am succesfully using within the DEFAULT.h file:

{0x10,0,217,1,-1,"Operation Mode"},
0x10,1,304,1,-1,"Defrost Operation"},
{0x20,0,105,2,1,"Outdoor air temp."},
{0x21,0,105,2,-1,"INV primary current (A)"},
{0x21,2,105,2,-1,"INV secondary current (A)"},
{0x30,0,152,1,-1,"INV frequency 1 (rps)"},
{0x30,1,152,1,-1,"INV frequency 2 (rps)"},
{0x30,0,211,1,-1,"Fan 1 (step)"},
{0x60,2,303,1,-1,"Thermostat ON/OFF"},
{0x60,2,302,1,-1,"Freeze Protection"},
{0x61,6,105,2,1,"Liquid temperature(R3T)"},
{0x61,8,105,2,1,"Inlet temperature(R4T)"},
{0x61,10,105,2,1,"DHW temperature(R5T)"},

neildsb, 2023-12-15:
> These are the ones i am succesfully using within the DEFAULT.h file:
>
> {0x10,0,217,1,-1,"Operation Mode"}, 0x10,1,304,1,-1,"Defrost Operation"}, {0x20,0,105,2,1,"Outdoor air temp."}, [...]

I have exposed a few more so I can calculate COP, here is my template if it helps?

    - name: "COP-power"
      unique_id: "espaltherma_cop"
      unit_of_measurement: 'COP'
      state: >
          {% if is_state_attr('sensor.esp_altherma','Operation Mode', 'Heating') and is_state_attr('sensor.esp_altherma','Freeze Protection', 'OFF')  %} 
          {{ 
          ((state_attr('sensor.esp_altherma','Flow sensor (l/min)')| float * 0.06 * 1.16 * (state_attr('sensor.esp_altherma','Leaving water temp. before BUH (R1T)') | float - state_attr('sensor.esp_altherma','Inlet water temp.(R4T)')|float) )
          /
          (states('sensor.heat_pump_power') | float /1000 ))
          |round(2)
          }}
          {% else %} 0 {%endif%}

youmustcomply, 2023-12-15:
Thats interesting, what is the unit thats being provided by the shelly?

I've so far been unsuccessful  in calculating the COP

neildsb, 2023-12-15:
> Thats interesting, what is the unit thats being provided by the shelly?
> 
> I've so far been unsuccessful in calculating the COP

Watts -

[image]

[image]

youmustcomply, 2023-12-19:
So I have created a new file under sensors which i keep all my sensor config files to try and keep it separate for each system.

The issue i am having is actually getting it to like the config. I have temporarily used my Victron reading for Watts as a test. This is everything I have in my ESP.yaml file:

  template:
    sensor:
      - name: "COP-power"
        unique_id: "espaltherma_cop"
        unit_of_measurement: 'COP'
        state: >
          {% if is_state_attr('sensor.none_althermasensors','Operation Mode', '>
          {{
          ((state_attr('sensor.none_althermasensors','Flow sensor (l/min)')| fl>
          /
          (states('sensor.victron_system_consumption_l1') | float /1000 ))
          |round(2)
          }}
          {% else %} 0 {%endif%}

Any ideas?

neildsb, 2023-12-19:
> So I have created a new file under sensors which i keep all my sensor config files to try and keep it separate for each system.
>
> The issue i am having is actually getting it to like the config. I [...]

it looks like you are missing loads of values, also check the naming of items

- name: "COP-power"
  unique_id: "espaltherma_cop"
  unit_of_measurement: 'COP'
  state: >
      {% if is_state_attr('sensor.esp_altherma','Operation Mode', 'Heating') and is_state_attr('sensor.esp_altherma','Freeze Protection', 'OFF')  %} 
      {{ 
      ((state_attr('sensor.esp_altherma','Flow sensor (l/min)')| float * 0.06 * 1.16 * (state_attr('sensor.esp_altherma','Leaving water temp. before BUH (R1T)') | float - state_attr('sensor.esp_altherma','Inlet water temp.(R4T)')|float) )
      /
      (states('sensor.heat_pump_power') | float /1000 ))
      |round(2)
      }}
      {% else %} 0 {%endif%}

youmustcomply, 2024-01-07:
Now that Xmas is out the way I actually have some time to play around. My copy and paste failed before and thats why it looked like it was missing loads of values. I have now successfully got it showing my COP within the developer tools and templates.

The part I am now struggling is where do i now put this. I have put it under my ESP config however it just gives me errors and about block start and block end.
[image]
[image]