raomin/ESPAltherma #348: DAIKIN EKHHP-A2V3

gomes89, 2023-11-17:
Hello. I see that my pump has the X10A connector with 8 pins. Is it supported? Which model should I select in the setup.h?

gomes89, 2023-11-18:
Update: I tested 1 or 2 includes so far and was only able to get data without errors from the PROTOCOL_S_ROTEX.

Sample data:

`{"Circulation pump":"OFF","External heater?":"OFF","Priority to domestic water":"OFF","Burner inhibit from solaris":"OFF","Refrig. Temp. liquid side(C)":46.1719,"Inlet water temp.(C)":212.953,"Outlet Water Temp.(C)":212.953,"D(C)":212.953,"DHW tank temp.(C)":44.1719,"F(C)":0.00390625,"Delta-Tr(deg)":0,"R/C Setpoint(C)":0,"Operation Mode":"Fan Only","Error Code":" 0","Thermo Off Error":" 0","Warning Code":" 0","Caution Code":" 0","????":-69.5547,"WifiRSSI":"-50dBm","FreeMem":"25800"}`

I would like to know a couple things more at least. For instance, the setpoint and if the HP is operating or not (with our without booster). Any ideas?

lougreenwood, 2024-07-03:
@gomes89 Did you make any progress on this? I have the same unit

gomes89, 2024-07-03:
> @gomes89 Did you make any progress on this? I have the same unit

Not really. I tested a bunch of the includes and this was the best I got.

lougreenwood, 2024-09-09:
@gomes89 I'm struggling with getting it working with my unit. I keep getting connection timeout errors.

Would you be able to share your `src/setup.h` (with your wifi commented out, etc), and your `include/def/PROTOCOL_S_ROTEX.h` so I can compare to my setup?

If you have any pics of your connection to hand that would be super useful too 😉 😁

gomes89, 2024-09-09:
> @gomes89 I'm struggling with getting it working with my unit. I keep getting connection timeout errors.
>
> Would you be able to share your `src/setup.h` (with your wifi commented out, etc), and your [...]

Unfortunately, I didn't take any pics and I'm currently away from the setup. In any way, the pinout is straightforward from the documentation. I would suggest you to re-check.

As for the `src/setup.h`, I didn't do anything custom. I just selected `#include "def/PROTOCOL_S_ROTEX.h"` and configured my own Wi-Fi, MQTT and GPIO PINs. Are your pins set correctly? Don't forget TX goes to RX in the pump, and vice-versa.

bagl3y, 2024-11-01:
Same here, I have an EKHHP-A2V3. The best I could get was with the PROTOCOL_S_ROTEX file. I don't know if we can get more information unfortunately.

I had a bit of trouble connecting my Dupont connectors because the connectors on the Daikin side are thinner, so I had to disassemble my Dupont and tighten them a little so that the contact was correct.

I didn't take a picture but it wasn't the most complex to find, the README is very well done for that (just, the picture of the 8pin connector in the readme is reversed compared to the position of the connector when the water heater is facing me.)

If anyone has any tips for having more data, I'm interested.

Here is the json I receive on the MQTT channel :

`{
   "Circulation pump":"OFF",
   "External heater?":"OFF",
   "Priority to domestic water":"OFF",
   "Burner inhibit from solaris":"OFF",
   "Refrig. Temp. liquid side(C)":21.2188,
   "Inlet water temp.(C)":212.953,
   "Outlet Water Temp.(C)":212.953,
   "D(C)":212.953,
   "DHW tank temp.(C)":48.2812,
   "F(C)":0.00390625,
   "Delta-Tr(deg)":0,
   "R/C Setpoint(C)":0,
   "Operation Mode":"Fan Only",
   "Error Code":" 0",
   "Thermo Off Error":" 0",
   "Warning Code":" 0",
   "Caution Code":" 0",
   "????":-69.5547,
   "M5BatV":"6.33V",
   "WifiRSSI":"-50dBm",
   "FreeMem":"204956"
}`