raomin/ESPAltherma #350: Altherma 3 R ECH2O

rob570, 2023-11-19:
Hi to everyone and Thank you Raomin for your project.
I've configured ESPAltherma for an Altherma 3 R ECH2O. I've used Top-Grade.h with two fixes.
I'm not able to find the registry for querying:

1. **mixed temperature (R7T)** the one found in other definitions don't work (always 0): {0x64,10,105,2,1,"Mixed water temp."},{0x64,10,118,2,1,"Mixed water temp.(R7T)"} 
2. **Tank Valve (M1S)**
3. **Bypass Valve (M2S)**

Here what I fixed in the description:

- {0x20,0,105,2,1,"Outdoor air temp.(R1T)"} should be {0x20,0,105,2,1,"Outdoor air temp.(R6T)"}
- {0x64,7,105,2,1,"Boiler Heating Target Temp."} it is the SH Water Heating Temp

Could anyone help me?
Thank you,
    Roberto
[image]

maromme, 2023-11-20:
I found those somewhere here in the discussions:
```
{0x63,8,105,2,1,"[HPSU] Mixed leaving water temperature after the tank R7T (DLWA2)"},
{0x63,10,101,1,-1,"[HPSU] Tank valve position (0:Bypass 100:Tank)"},
{0x63,11,101,1,-1,"[HPSU] Bypass valve position (0:Bypass 100:Emitter)"},
{0x20,0,105,2,1,"R6T"},
```

lfrancke, 2023-11-20:
Maybe these are helpful for you as well: [ech2o.csv](https://github.com/raomin/ESPAltherma/files/13419187/ech2o.csv)

rob570, 2023-11-21:
Dear Maromme and Ifrancke,
thank you for helping and sharing.
I'll try next week and I'll give you a feedback.
Have a nice day,
  Roberto

rob570, 2023-11-26:
Hi, I can confirm the settings provided work!
Thank you for your reply.
I attach a new definition file (renamed to csv) based on your information.
[ALTHERMA(ECH2O).h.csv](https://github.com/raomin/ESPAltherma/files/13466100/ALTHERMA.ECH2O.h.csv)
   Roberto

simonebortolin, 2024-01-01:
@lfrancke @rob570 why 
`{0x62,11,405,1,1,"Water pressure"},`
and not 
`{0x62,11,105,1,1,"Water pressure"},`

the conversion 405 is only for the R32 gas inside the refrigerant circuit.

rob570, 2024-01-01:
Thank you Simone,
That's why even after pulling the breaks on 405 conv ( https://github.com/raomin/ESPAltherma/pull/325 ) I've got wrong values.
Grazie!

simonebortolin, 2024-01-01:
how many Italians with this heat pump!

simonebortolin, 2024-01-01:
I am only missing the two 3-way valve values...

[image]
[image]

danilo-quadrelli, 2024-01-02:
Hi, I can't confirm the settings provided work!
In my case:
{0x62,11,405,1,1,"Pressione acqua"}, return 
Pressione acqua: -51.1813  (Daikin 1,3 Bar)

{0x62,11,105,1,1,"Pressione acqua"}, return 
Pressione acqua: 0 (Daikin 1,3 bar)

and:

{0x63,10,101,1,-1,"[HPSU] Tank valve position (0:Bypass 100:Tank)"},
{0x63,11,101,1,-1,"[HPSU] Bypass valve position (0:Bypass 100:Emitter)"},

[HPSU] Tank valve position (0:Bypass 100:Tank): 0
[HPSU] Bypass valve position (0:Bypass 100:Emitter): 0   (Daikin 12%)

Thank you

simonebortolin, 2024-01-02:
```
In my case:
{0x62,11,405,1,1,"Pressione acqua"}, return
Pressione acqua: -51.1813 (Daikin 1,3 Bar)

{0x62,11,105,1,1,"Pressione acqua"}, return
Pressione acqua: 0 (Daikin 1,3 bar)
```

Molto strano, a me torna il valore corretto
```
{0x63,10,101,1,-1,"[HPSU] Tank valve position (0:Bypass 100:Tank)"},
{0x63,11,101,1,-1,"[HPSU] Bypass valve position (0:Bypass 100:Emitter)"},
```

pure queste a me tornano corretti i valori, che modello di ECH2O hai?

danilo-quadrelli, 2024-01-02:
Daikin Altherma 3 R ECH2O:

EHSXB08P30DA

simonebortolin, 2024-01-02:
vecchia elettronica, potrebbe essere molto diverso da quelli dalla serie nuova con eye