raomin/ESPAltherma #367: Leaving Water before BUH RT1 vs AFTER BUH RT2 - CoP calculation descrepency

stephenacrown, 2023-12-11:
On the main page "how to calculate CoP, it is stated in the formula to use Leaving Water before BUH RT1 (as that is common across Daikin models it would seem. 

However, for those with the backup heater (BUH), and even if not actively used, we are seeing a large temperature drop on the sensor after this stage:

Leaving Water after BUH RT2

[image]

There is a discussion over on OpenEnergyMonitor where someone with a MID billing grade heat meter is suggesting the RT1 sensor is not appropriate for use in CoP calculations, and that RT2 more closely matches their heat meter values. 

Is anyone else aware of why there is such a large temperature drop (0.8-1.8C) have been seen between stages, which feels "off" given the heat loss that would infer. 

https://community.openenergymonitor.org/t/daikin-9kw-ashp-application-discussion/24848/101?u=stephen_crown

stephenacrown, 2023-12-11:
Forgot to add an example:

[image]

rufik, 2024-01-13:
Well, really interesting. I've created COP template sensors for R1T and for R2T, so I'm going to follow the both numbers.

JKGBIT, 2024-01-21:
👍 I also have seen this, and switched my own code to R2T, disapointed im not actually getting a cop of 8.9 lol. As the old saying goes, "_if it seems to good to be true, its probably a duck_" - something like that anyway. 4.8 Cop is okay too I suppose. 👍

rufik, 2024-01-21:
Using R2T results in about 10% lower COP for me. It's just rough math here, I need to create a template sensor as % of difference I think...

JKGBIT, 2024-01-22:
I expect that R1T Is very close to the heat exchanger and will be picking up on the emitted heat from that and this hotter than the water inside it. The controller probably uses that to check the heat exchanger is working outputting heat normally and will kick in the backup heater if required and allowed by settings. 
And probably not a reliable source of data for cop. 
Mine is 2.5° off R2T.

jonputsman, 2024-01-22:
While I agree that R1T looks too good to be true for use in COP calculations, I tend to see R1T being used to match the leaving water temperature set point:

[image]

I’m going to start logging flow and return inside my property, as I think my system is delivering a wider flow-return delta than I’ve requested.

JKGBIT, 2024-01-22:
Interesting. I think I saw someone on forum put a request into Daikin, so wonder what they said about it.

jonputsman, 2024-01-22:
That could have been my post. I’ve heard nothing significant from Daikin regarding the various temperature probes.

stephenacrown, 2024-01-22:
> Interesting. I think I saw someone on forum put a request into Daikin, so wonder what they said about it. 

Here is a screenshot of the Daikin response to Matt's question on this point:

[image]

See here for the Daikin response to Matt:

https://community.openenergymonitor.org/t/daikin-9kw-ashp-application-discussion/24848/193?u=stephen_crown

I will also add, that models without a BUH, the R1T sensor is actually int he same position as ours with a BUH. 

[image]

[image]

rufik, 2024-01-23:
I've made a "COP template sensor" using R1T oraz second one using R2T temperatures. Than I've made a sensor with difference between these two COPs.
My LT EHVH DA6V (8kW) has a BUH (4+2kW) and the diff between COPs like like that:
[image]

As you can see COP R2T is a little but higher than COP R1T :) The average diff is about 0.4°C, maybe 0.5°C. The percentage diff is about 10%:
[image]

I use ESPAlthera software from 2021.12.06 with def a little bit tuned, but the essential sensors are:
```
LabelDef PROGMEM labelDefs[] = {
	{0x21,0,105,2,-1,"INV primary current (A)"},
	{0x21,4,101,2,-1,"Voltage (N-phase) (V)"},
	{0x61,8,105,2,1,"Inlet water temp.(R4T)"},
	{0x61,2,105,2,1,"Leaving water temp. before BUH (R1T)"},
	{0x61,4,105,2,1,"Leaving water temp. after BUH (R2T)"},
	{0x62,9,105,2,-1,"Flow sensor (l/min)"},
}
```

Is COP diff really that small (~10%) for my HP model or should I update ESPAltherma software because of some bugs fixed? :)

johngouk, 2024-03-06:
I have an EDLA09D3V3 with BUH, and I had a significant discrepancy between R1T and R2T, of around 2C, and R2T and R4T of 1.5C. However, after I told Daikin support about this, I had an engineer visit and remove a link on the tape heater power block X7M 5-7. The tape heaters E11H and E12H, on the plate heat exchanger outlet and inlet respectively, which had previously been on whenever the circulation pump was running, stayed off, as they should according to the service manual! The R1T, R2T and R4T readings are now within 0.1C of each other. So it's worth checking this - there is a technical service bulletin on it apparently.

raomin, 2026-10-23:
Closing with the 2.0 release: the doc now says which sensor to use (R1T, or R2T when your meter also powers the backup heater) and why. Thanks for the discussion, it shaped that paragraph.

closed 2026-10-23 by raomin

raomin, 2026-11-03:
For those who missed it, I made an analysis on the R1T vs R2T on the boards running v2. See [here](https://github.com/raomin/ESPAltherma/discussions/599#discussioncomment-19354456)