raomin/ESPAltherma #369: Daikin Altherma MT: Floor Heating Mixer Position value

habitoti, 2023-12-18:
Hi,
is anyone aware of a parameter set that would deliver the opening percentage of the floor heating mixer valve? It is shown in the info panel of the indoor unit, so it is read properly from the separate mixing station. However I haven't found anything in the setup file that would be provided by ESPAltherma.
Thanks, habitoti

raomin, 2024-01-07:
In my case, (EPGA / EAV), it's `[Future] 3 way Valve Mixing 1` but as it's written, it's not supported yet... :(
Maybe it is in yours?

raomin, 2026-10-23:
@habitoti It's been a while. I checked while drawing the two-zone units: the mixing valve opening is not in any registry on the wire, only the mixed water temperature and the pumps. Closing as not available.

closed 2026-10-23 by raomin