raomin/ESPAltherma #370: Daikin Altherma MT: indoor temperature always 20°C

habitoti, 2023-12-18:
The indoor temperature is constantly reported as 20°C, which is definitely not correct (or sometimes its is, though ☺️). Not that I really need that data, however I just wondered whether that is something only I am seeing or some other "hidden" problem behind that (and maybe other data is off or not current, too, then?)

simonebortolin, 2024-01-01:
do you have the internal probe and/or madoka?

habitoti, 2024-01-02:
No, don‘t have a Madoka. What would this reading be? I thought the temperature of the room  where the indoor unit is located (while Madoka would deliver the temperature of a heated room…)? Maybe I was just confused by the constant 20 degrees — would have expected 0 if it would be the reading of some external probe…

simonebortolin, 2024-01-02:
is simply the default value in the absence of a makoda probe or probe

raomin, 2024-01-06:
That clears it. Thanks @simonebortolin

closed 2024-01-06 by raomin