raomin/ESPAltherma #377: Daikin Altherma 2 - Law of water default

MathieuLH44, 2024-01-10:
Hello,

I use ESPAltherma since this summer, but, about 1 month ago, when the temperature was cold (0°C), the law water didn't agree with the programmation.

[image]

I want 45°C at -3°C and 35 at 15°C

When the law of water is is default, the pump start and stop often

[image]

This problem is when the outside temperature is very low, and when I use the external thermostat

Someone had the same issue ?

MathieuLH44, 2024-01-12:
I have 2 law of water, 1 principal and 1 added, but the values are differents
Do you know what mean each one ?

MathieuLH44, 2024-01-13:
Today, I tried to use the internal thermostat, and the water law, it worked.
After, I tried to use the external thermostat and the absolute temperature, and, after about 1 hour, I have the same issue

Someone else have this issue ?

THK56, 2024-01-18:
Maybe that are the defrost cycles. Defrosting is necessary at temperatures below approx 2-4°C. Defrosting means, that the heat pump "reverses" the operation and heats up the evaporator (the radiator of the outdoor unit). That takes some minutes, then the heat pump reverses again. So check the state of the sensor "Dégivrage" or "Defrost Operation" during the drop of leaving temparature. e.g. my screenshot of leaving temp and defrost
[image]

THK56, 2024-01-19:
Do you use Room thermostat control (= the system controls the room temp and the LWT, versus the system controls the LWT according to a schedule/fix/outdoor temp dependant) ? And do you have the issue with different behaviour depending of using internal or external room thermostat? And what do you mean, when writing "using external remote" vs "internal remote"? Is remote the room thermostat?
A short side comment: room thermostat control never worked well with my pump. I had field service people of the dealer and remote contact with daikin. The room temp simply oscillated approx. 2-3°C in the room. Daikin suggested to use LWT control in my situation.

THK56, 2024-01-19:
There are different parameters that control the (water) pump. See field service installation manual.
Be aware that these parameters may vary in your case!!
[0F-0D] operation mode of the pump
[0F-01] temp limit for cooling
[04-02] temp limit for heating
[0F-00] operation limitation depending on outdoor temp
[0F-09] pump operation in case of flow error
[09-0D] limitation of pump flow
[05-0D] operation depending on electricity supplier's rules

MathieuLH44, 2024-01-19:
I will check these parameters

The first time that I reach the desired temperatured, all work well, but, since the second time, the LWT stay too cold (currently, around 36°C but wanted LWT is about 45°C)

THK56, 2024-01-19:
Check the parameters of the heating curve as well [01-00 .. -03] as well. Just to be sure

MathieuLH44, 2024-01-19:
[image]

[image]

I check these values, I seems me right

bashers222, 2024-01-30:
This is most defintely the way Daikin handle cool weather, and not an issue as such
At 8 degrees, seemingly regardless of antifreeze presence setting, the pump moves the fluid around the system in order to check the real flow temperatures in the pipes and limit the chance of freezing.
Even if it needs to warm the pipes to 35 degrees, its pretty quick and does not cost much

[image]

[image]

[image]

MathieuLH44, 2024-02-04:
ok, because this happen only when I use the external thermostat