raomin/ESPAltherma #381: Run on M5Stack ESP32 Ethernet Unit with PoE

abbottcam, 2024-01-14:
Hi there, ive been using this project on the M5 StickPlus with no problems, apart from the wifi being a bit flaky. I was therefore wondering if it would run on M5Stack ESP32 Ethernet Unit with PoE?

Thanks

James

Xylaw, 2024-02-02:
Hi, I would also like to know if and how to get this stick to work via ethernet.
Thanks!

swalsh11, 2024-02-09:
Hi, I bought one, as it It looked like it can be flashed, and i may be able to use G16 & G17 to talk to x10, See this post:  https://community.m5stack.com/topic/4645/how-to-connect-to-m5stack-poesp32, However the M5 Stack USB ttl is out of stock, and a Standard CP210x USB to TTL appears to be not working.  I will see if i can get the correct TTL, Otherwise I need to figure out how to Enable Serial UART from the M5 stick to the POESP32, and use the Grove port Connection as intended,  Not sure how to proceed on this.

Alibloke, 2024-02-14:
You can flash with a USB to TTL adapter by bridging G0 to ground

swalsh11, 2024-02-14:
Thanks, I will try that

Alibloke, 2024-02-14:
Some code changes will be needed to remove wifi and add ethernet:
https://github.com/raomin/ESPAltherma/discussions/366