raomin/ESPAltherma #390: How to get valid readings from Externally mounted Thermistor R6T

bashers222, 2024-01-30:
Im trying to activate the more accurate external thermistor attached to R6T on the Dakin

When i uncomment that line: "Ext. indoor ambient sensor (R6T)" in the defintion, the reading that is returned is -128

Do i need to configuire it diferently for this thermistor?

THK56, 2024-02-13:
yes needs configuration, depending on your model, probably [0C-08]=2. pls check first.

bashers222, 2024-02-13:
The thermistor is configured in the heat pump as it's showing a reading in the sensors page.
It just comes out as -128 in espaltherma

THK56, 2024-02-13:
that needs some debugging. uncomment the print-statement in the conversion 105 in file "converters.h" (line 90):
        case 105:
            dblData = (double)getSignedValue(data, num, 0) * 0.1;
            // Serial.printf("%f\n", dblData);   <----- uncomment this line
            break;
then build, upload aso.

THK56, 2024-02-13:
but: there are different values, that are converted with "case 105". so you will get diff values printed out, depending what other values you also use. so to be sure you get your external temp sensor value, comment all others with conversion 105 temporary out.

bashers222, 2024-02-13:
Ok. Will try that later if I get a chance

Regards
Ashley Slade

TRIROG, 2024-02-25:
@bashers222  did you maybe resolve this?

bashers222, 2024-02-25:
as my production ESP unit does not has OTA upload enabled i tried flashing the code to another esp8266 but got corrupted
I will attempt this later this coming week with a wired EPS32 as i need to get a more stable connection than i have currently

bashers222, 2024-02-25:
I do appreciate the pointers you supplied and i will feed back on my findings

TRIROG, 2024-02-25:
Great! Would you share unit model?

bashers222, 2024-02-25:
Its a EDLQ05CV3 monoblock unit and i'm using the definition for "Altherma(LT_EBLQ-EBLQ-CA series 5-7kW Monobloc"

bashers222, 2024-03-14:
@THK56 I've made the changes as suggested
Uncomented the "Serial.print" and also removed everything out of my def apart from the required element

The MQTT output is:
`{"Outdoor air temp.(R1T)":13.5,"Ext. indoor ambient sensor (R6T)":-128,"WifiRSSI":"0dBm","FreeMem":"248868"}`

But that value is not varying
The current outdoor temperature is 11.4C as reported on the panel

THK56, 2024-03-20:
Then check:
- what def file do you use? I'm wondering that in some def files the R6T is on 0x61,14,105,2,1,.... and in some others it's on 0x61,13,105,2,1,... e.g. EGSAH-X-EWSAH-X-D. Pos 13 seems wrong to me. because the preceding value is on pos 12 and needs 2 bytes --> 12 + 2 = 14

bashers222, 2024-03-22:
I'm using "Altherma(LT_EBLQ-EBLQ-CA series 5-7kW Monobloc)"

The original value is 14, but I changed that value to 13 and the returned value was 0