raomin/ESPAltherma #392: Wrong pinout description for 8 pin X10A Connection

mschaeuble, 2024-02-04:
In my case of a Rotex RKHBH016BB9WN, the pinout description of the 8 pin X10A connection was wrong. Correct is:

1: 5V (missing in the description)
3: TX (not RX as in the description)
4: RX (not TX as in the description)
8: GND

I don't know whether this pinout is valid for all devices, or just the older Rotex devices. In any case, the Readmy at https://github.com/raomin/ESPAltherma?tab=readme-ov-file#8-pin-x10a-connection should be updated/extended.

raomin, 2024-03-12:
Don't be offended by the question but: do you know that RX (receiver side) are to be connected on the TX (emitter side) and vice versa? So on the board side the RX pin is to be connected on the TX side of the esp32/serial port reader.

If you know and that is not a mistake, then they might have changed the pinout, albeit surprising...
I'll wait for your confirmation before updating the page.

mschaeuble, 2024-03-14:
@raomin Thanks a lot for your hint. I checked my wiring and you're right: By just looking at the "8 pin X10A Connection" section, I didn't realize that the wiring should be RX<->TX and vice versa. Therefore, I can confirm that the pinout description is correct.

Still, it might be helpful for some people to indicate that pin 1 of the 8 pin X10A connector delivers +5V, which can be used as the supply voltage for the ESP.

raomin, 2024-03-21:
Hmmm, it's  in the table right above, it's also shown in the diagram and explained in the text twice. I guess it's enough...
Sorry @mschaeuble, but would you have a tendency to rush into conclusions? ;)

closed 2024-03-21 by raomin

mschaeuble, 2024-03-21:
@raomin I don't think offending people will help in any project, especially open-source projects.  

In my opinion, there is no word/diagram/table describing the +5V output of the **8-pin** connector. The text/diagram/table refers to the **4-pin** connector only - the 8-pin connector "connects differently" according to your words.

raomin, 2024-03-21:
Hi @mschaeuble, I owe you some apologies. First I didn't mean to offend you and second, I thought you were referring to the 5 pin connector, looks like I'm the one to blame for rushing into conclusions. ;)
I updated the description. Thanks.