raomin/ESPAltherma #396: M5stickc Plus 2 problems

luca320, 2024-02-14:
Ciao a tutti. Premetto che utilizzo da anni una m5stickc, che ho programmato con il software visual studio. Non ho avuto alcun problema. Letture precise di tutti i sensori espaltherma. Ho comprato da poco una nuova m5stickc plus 2, che ho pensato di tenere di scorta nel caso la prima chiavetta dovesse non funzionare più.  Ho formattato la m5stickcplus usando m5burner e uniflow dedicato al modello corretto, ho fatto upload del progetto selezionando la porta com corretta e come modello, non essendoci la plus 2, ho messo m5stickc plus. la procedura ha esito positivo, non ci sono errori e la scrittura sulla chiavetta va a buon fine. però non riesco a vederla online all'interno dei dispositivi collegati al router. Quale può essere il problema? Ringrazio chi si prenderà il tempo di rispondermi.

Hi everyone. Let me start by saying that I have been using an m5stickc for years, which I programmed with the visual studio software. I didn't have any problems. Precise readings of all espaltherma sensors. I recently bought a new m5stickc plus 2, which I thought I'd keep as a spare in case the first stick no longer works. I formatted the m5stickcplus using m5burner and uniflow dedicated to the correct model, I uploaded the project by selecting the correct com port and as the model, since there was no plus 2, I put m5stickc plus. the procedure is successful, there are no errors and the writing on the flash drive is successful. but I can't see it online within the devices connected to the router. What could be the problem? I thank those who take the time to answer me.

rkontrimas, 2024-02-17:
I am new user of this project and I'm also having same problem. Also, screen is not showing any information after uploading, which I believe i should.
Following this issue

MarkLofthouse, 2024-02-23:
Same here - found another thread that says the code doesn’t work on the Plus 2, can we get a view if it’s in the roadmap before I return the plus 2? Thanks.

raomin, 2024-02-24:
I'll have a look.
Please open the serial monitor and paste the logs from the device.

MarkLofthouse, 2024-02-24:
> I'll have a look. Please open the serial monitor and paste the logs from the device.

Thanks
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4

My little bit of research says that M5StickC PLUS2 is an iterative version of M5StickC PLUS, featuring the ESP32-PICO-V3-02 chip, I also found someone saying on a forum the display pinout was different.

Thanks for looking.

raomin, 2024-02-24:
Looks like there is no program on your device.
Can you paste the upload process?

MarkLofthouse, 2024-02-24:
*  Executing task: platformio run --target upload --environment m5stickcplus --upload-port /dev/cu.usbserial-568A0021941 

Processing m5stickcplus (platform: espressif32; board: m5stick-c; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stick-c.html
PLATFORM: Espressif 32 (6.5.0) > M5Stick-C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- M5StickCPlus @ 0.1.0
|-- PubSubClient @ 2.8.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- EEPROM @ 2.0.0
Building in release mode
Retrieving maximum program size .pio/build/m5stickcplus/firmware.elf
Checking size .pio/build/m5stickcplus/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.2% (used 56260 bytes from 327680 bytes)
Flash: [=======   ]  65.8% (used 862801 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: /dev/cu.usbserial-568A0021941
Uploading .pio/build/m5stickcplus/firmware.bin
esptool.py v4.5.1
Serial port /dev/cu.usbserial-568A0021941
Connecting..........
Chip is ESP32-PICO-V3-02 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, Embedded PSRAM, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e8:9f:6d:31:3f:54
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000e4fff...
Compressed 17536 bytes to 12202...
Writing at 0x00001000... (100 %)
Wrote 17536 bytes (12202 compressed) at 0x00001000 in 1.5 seconds (effective 92.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 335.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 455.9 kbit/s)...
Hash of data verified.
Compressed 869104 bytes to 554088...
Writing at 0x00010000... (2 %)
Writing at 0x00019d94... (5 %)
Writing at 0x0002589f... (8 %)
Writing at 0x000323f2... (11 %)
Writing at 0x0003d9c4... (14 %)
Writing at 0x0004371e... (17 %)
Writing at 0x00048c39... (20 %)
Writing at 0x0004e501... (23 %)
Writing at 0x00053ae6... (26 %)
Writing at 0x0005911c... (29 %)
Writing at 0x0005e28a... (32 %)
Writing at 0x00063553... (35 %)
Writing at 0x00068821... (38 %)
Writing at 0x0006da98... (41 %)
Writing at 0x00072d91... (44 %)
Writing at 0x000781a6... (47 %)
Writing at 0x0007d42e... (50 %)
Writing at 0x00082fb0... (52 %)
Writing at 0x000887a4... (55 %)
Writing at 0x0008db36... (58 %)
Writing at 0x00092e6c... (61 %)
Writing at 0x00098255... (64 %)
Writing at 0x0009d870... (67 %)
Writing at 0x000a2e8d... (70 %)
Writing at 0x000a8635... (73 %)
Writing at 0x000ae592... (76 %)
Writing at 0x000b3eb9... (79 %)
Writing at 0x000b9819... (82 %)
Writing at 0x000be8ec... (85 %)
Writing at 0x000c6f39... (88 %)
Writing at 0x000cf4bd... (91 %)
Writing at 0x000d4613... (94 %)
Writing at 0x000da10e... (97 %)
Writing at 0x000df580... (100 %)
Wrote 869104 bytes (554088 compressed) at 0x00010000 in 53.4 seconds (effective 130.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
====================================================== [SUCCESS] Took 62.23 seconds ======================================================

Environment    Status    Duration
-------------  --------  ------------
m5stickcplus   SUCCESS   00:01:02.234
====================================================== 1 succeeded in 00:01:02.234 ======================================================
 *  Terminal will be reused by tasks, press any key to close it.

smart40-githib, 2024-02-25:
I've made a fork and added support for M5Stick Plus2
https://github.com/smart40-githib/ESPAltherma

It seems to work fine  so far as I can tell as I won't be opening the heatpump until the summer.

There is a pull request to merge into this repo.

MarkLofthouse, 2024-02-25:
> I've made a fork and added support for M5Stick Plus2 https://github.com/smart40-githib/ESPAltherma
> 
> It seems to work fine so far as I can tell as I won't be opening the heatpump until the summer.
> 
> There is a pull request to merge into this repo.

Thank - sorry noob here.

I've downloaded your fork, and pushed to my plus 2, but nothing different. What should the Env: be set to?

Thanks for your help.

rkontrimas, 2024-02-25:
@smart40-githib for me it also seem to work now, I see 2 new entities in HA(AlthermaSensor and AlthermaSwitch). I will try to connect to my heatpump tomorrow. Thanks!

```
--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Querying register 0x20... Time out! Check connection
Querying register 0x21... Time out! Check connection
Retrying...
Querying register 0x21... Time out! Check connection
Retrying...
Querying register 0x21... Time out! Check connection
Retrying...
```

MarkLofthouse, 2024-02-25:
Same here - thank you.

Will this get put into the main repo next?

MarkLofthouse, 2024-02-26:
Morning, I can see an entity - sensor.none_althermasensors, however, it has no attributes, other than 'Online'.

I've uninstalled and re-installed MQTT - and get the same issues.

However, if I listen to the topics in MQTT, I can see all the data coming from the heatpump.

[image]

MarkLofthouse, 2024-02-26:
Might have found my problem - although not sure how to fix it.

[image]

smart40-githib, 2024-02-26:
Fixed the syntax error and battery voltage value.
Can you try again please.

MarkLofthouse, 2024-02-27:
> Fixed the syntax error and battery voltage value. Can you try again please.

Thanks, just copied mqtt.h across to my deployment, and seems to be working fine.

luca320, 2024-03-01:
It seems to work and for that I am very grateful. However, the black screen problem persists. I don't understand why. I made this request for help for a friend. Now its works perfectly and the screen works too. In my case however the screen remains black. This also happened to me on the M5stickcplus, but the latter still works perfectly (I don't care much about having a working screen). I was hoping that in the m5stickcplus 2 the screen would work.

luca320, 2024-03-01:
> I am new user of this project and I'm also having same problem. Also, screen is not showing any information after uploading, which I believe i should. Following this issue

Have you solved the black screen problem?

smart40-githib, 2024-03-01:
The should come on and then turn off after 30 seconds. If you press the button on the front it should come on for another 30 seconds. Other than that I cannot help.

raomin, 2024-03-21:
No answer? I guess it's solved by now.

closed 2024-03-21 by raomin