raomin/ESPAltherma #408: Build fails for M5stickc Plus 2

vdiogo, 2024-03-12:
I have followed the instructions on the Git (first time I have used Visual Studio Code or M5stack products) and after downloading the entire code to my local disk and opening it with VSC, adjusted:
- MQTT settings
- WiFi
- Altherma model
- Sensors

Tried with environment m5stickcplus2 (later changed to m5stickcplus) , connected the M5StickC Plus2 to macOs and pressed PlatforIO:Upload, and the process failed.

Afterwards, I tried compiling the code (PlatforIO:Build) and it also fails.

I can't determine if this is related with the ESP32 used, or some other issue with the compiled code. Happy to further troubleshoot if someone can guide me here.

I did found some errors:
```
Processing m5stickcplus (platform: espressif32; board: m5stick-c; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stick-c.html
PLATFORM: Espressif 32 (6.5.0) > M5Stick-C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- M5StickCPlus @ 0.1.0
|-- PubSubClient @ 2.8.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- EEPROM @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
Building in release mode
Compiling .pio/build/m5stickcplus/src/main.cpp.o
Compiling .pio/build/m5stickcplus/lib849/M5StickCPlus/utility/Speaker.cpp.o
Compiling .pio/build/m5stickcplus/lib849/M5StickCPlus/utility/Sprite.cpp.o
Compiling .pio/build/m5stickcplus/lib849/M5StickCPlus/utility/qrcode.c.o
src/main.cpp: In function 'void extraLoop()':
src/main.cpp:100:8: error: 'class M5StickCPlus' has no member named 'Display'
     M5.Display.wakeup();
        ^~~~~~~
src/main.cpp:103:8: error: 'class M5StickCPlus' has no member named 'Display'
     M5.Display.sleep();
        ^~~~~~~
src/main.cpp: In function 'void setupScreen()':
src/main.cpp:256:10: error: 'class M5Display' has no member named 'setBrightness'
   M5.Lcd.setBrightness(127);
          ^~~~~~~~~~~~~
src/main.cpp:258:10: error: 'class M5Display' has no member named 'setFont'; did you mean 'textfont'?
   M5.Lcd.setFont(&FreeSansBold12pt7b);
          ^~~~~~~
          textfont
src/main.cpp:266:10: error: 'class M5Display' has no member named 'setFont'; did you mean 'textfont'?
   M5.Lcd.setFont(&Font0);
          ^~~~~~~
          textfont
src/main.cpp:266:19: error: 'Font0' was not declared in this scope
   M5.Lcd.setFont(&Font0);
                   ^~~~~
src/main.cpp:266:19: note: suggested alternative: 'font'
   M5.Lcd.setFont(&Font0);
                   ^~~~~
                   font
*** [.pio/build/m5stickcplus/src/main.cpp.o] Error 1
======================================================== [FAILED] Took 5.63 seconds ========================================================

Environment    Status    Duration
-------------  --------  ------------
m5stickcplus   FAILED    00:00:05.627
================================================== 1 failed, 0 succeeded in 00:00:05.627 ==================================================

 *  The terminal process "platformio 'run', '--target', 'upload', '--environment', 'm5stickcplus', '--upload-port', '/dev/cu.usbmodem575C0307011'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

```

vdiogo, 2024-03-12:
I was able to compile the code (commented some lines of code) so I am now down to not being able to install it:

```
Processing m5stickcplus2 (platform: espressif32; board: m5stick-c; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stick-c.html
PLATFORM: Espressif 32 (6.5.0) > M5Stick-C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- M5StickCPlus2 @ 1.0.2
|-- PubSubClient @ 2.8.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- EEPROM @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
Building in release mode
Retrieving maximum program size .pio/build/m5stickcplus2/firmware.elf
Checking size .pio/build/m5stickcplus2/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.1% (used 62556 bytes from 327680 bytes)
Flash: [=======   ]  74.4% (used 975065 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: /dev/cu.usbmodem575C0307011
Uploading .pio/build/m5stickcplus2/firmware.bin
esptool.py v4.5.1
Serial port /dev/cu.usbmodem575C0307011
Connecting.....
Chip is ESP32-PICO-V3-02 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, Embedded PSRAM, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 10:06:1c:17:06:c8
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
*** [upload] Error 2
======================================================= [FAILED] Took 9.38 seconds =======================================================

Environment    Status    Duration
-------------  --------  ------------
m5stickcplus2  FAILED    00:00:09.383
================================================== 1 failed, 0 succeeded in 00:00:09.383 ==================================================

 *  The terminal process "platformio 'run', '--target', 'upload', '--environment', 'm5stickcplus2', '--upload-port', '/dev/cu.usbmodem575C0307011'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

vdiogo, 2024-03-12:
Also did the following:
- Make sure that nothing is connected on the board when you upload. `checked`
- Try to lower your upload speed upload_speed = 115200 in your platformio.ini: `done`
- Press the reset button and release just before the upload starts. `doesn't have reset button, but disconnected/ connected before flashing`

raomin, 2024-03-12:
Yes, you eventually chose the right environment.
You have a small button on the side normally, let it press 7 sec to start the esp and upload again.

vdiogo, 2024-03-13:
Reverted to the first version of teh code (can compile) and tried flashing on a regular ES32 and get same error `Operation timed out`:
```
Processing esp32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.5.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- PubSubClient @ 2.8.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- EEPROM @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
Building in release mode
Retrieving maximum program size .pio/build/esp32/firmware.elf
Checking size .pio/build/esp32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.0% (used 59092 bytes from 327680 bytes)
Flash: [======    ]  61.4% (used 804929 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: /dev/cu.usbmodem57160279681
Uploading .pio/build/esp32/firmware.bin
esptool.py v4.5.1
Serial port /dev/cu.usbmodem57160279681
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: fc:b4:67:f5:6e:bc
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
*** [upload] Error 2
======================================================= [FAILED] Took 4.88 seconds =======================================================

Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:04.882
================================================== 1 failed, 0 succeeded in 00:00:04.882 ==================================================

 *  The terminal process "platformio 'run', '--target', 'upload', '--environment', 'esp32', '--upload-port', '/dev/cu.usbmodem57160279681'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

vdiogo, 2024-03-15:
I created a build from a forked environment and managed to compile and upload the build. Snag is that I had to select the `M5stickc Plus` environment as the `M5stickc Plus 2` was not available to select.
I'll wait fro the new code to be available as per https://github.com/raomin/ESPAltherma/issues/410#issuecomment-1999015789 and try again

raomin, 2024-03-16:
I fixed the issue with `PIN_THERM_ACTIVE_STATE` not being available, just committed it.
Did you manage to get your upload working, do you know what it was?

vdiogo, 2024-03-16:
> I fixed the issue with `PIN_THERM_ACTIVE_STATE` not being available, just committed it. Did you manage to get your upload working, do you know what it was?

I managed to upload the code from the Git fork (no errors thrown) but nothing appears in the lcd (stays black) and no mqtt report (ps: still not connected to the heat pump) 

Still getting errors with new code:
```
Processing m5stickcplus2 (platform: espressif32; board: m5stick-c; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stick-c.html
PLATFORM: Espressif 32 (6.5.0) > M5Stick-C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- M5StickCPlus2 @ 1.0.2
|-- PubSubClient @ 2.8.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- EEPROM @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
Building in release mode
Compiling .pio/build/m5stickcplus2/src/main.cpp.o
Building .pio/build/m5stickcplus2/bootloader.bin
Generating partitions .pio/build/m5stickcplus2/partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio/build/m5stickcplus2/lib03e/SPI/SPI.cpp.o
Compiling .pio/build/m5stickcplus2/lib329/Wire/Wire.cpp.o
Compiling .pio/build/m5stickcplus2/libbeb/M5GFX/M5GFX.cpp.o
Compiling .pio/build/m5stickcplus2/libbeb/M5GFX/lgfx/Fonts/IPA/lgfx_font_japan.c.o
Compiling .pio/build/m5stickcplus2/libbeb/M5GFX/lgfx/Fonts/efont/lgfx_efont_cn.c.o
Compiling .pio/build/m5stickcplus2/libbeb/M5GFX/lgfx/Fonts/efont/lgfx_efont_ja.c.o
Compiling .pio/build/m5stickcplus2/libbeb/M5GFX/lgfx/Fonts/efont/lgfx_efont_kr.c.o
In file included from src/main.cpp:31:
include/mqtt.h: In function 'void callbackSafety(byte*, unsigned int)':
include/mqtt.h:212:37: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:212:37: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
                                     SG_RELAY_ACTIVE_STATE
include/mqtt.h:218:36: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:218:36: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
                                    SG_RELAY_ACTIVE_STATE
src/main.cpp: In function 'void setup()':
src/main.cpp:314:35: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:314:35: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
                                   SG_RELAY_ACTIVE_STATE
Archiving .pio/build/m5stickcplus2/lib03e/libSPI.a
Archiving .pio/build/m5stickcplus2/lib329/libWire.a
*** [.pio/build/m5stickcplus2/src/main.cpp.o] Error 1
Indexing .pio/build/m5stickcplus2/lib03e/libSPI.a
Indexing .pio/build/m5stickcplus2/lib329/libWire.a
======================================================= [FAILED] Took 7.19 seconds =======================================================

Environment    Status    Duration
-------------  --------  ------------
m5stickcplus2  FAILED    00:00:07.194
================================================== 1 failed, 0 succeeded in 00:00:07.194 ==================================================

 *  The terminal process "platformio 'run', '--environment', 'm5stickcplus2'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

TinusDee, 2024-03-17:
Same issue here, I'm also running VSC with platform on a Mac and unable to flash/upload the M5stickC. Getting the same error code.

Processing m5stickc (platform: espressif32; board: m5stick-c; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stick-c.html
PLATFORM: Espressif 32 (6.1.0) > M5Stick-C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20007.0 (2.0.7) 
 - tool-esptoolpy @ 1.40500.0 (4.5.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- M5StickC @ 0.2.5
|-- PubSubClient @ 2.8.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- EEPROM @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
Building in release mode
Compiling .pio/build/m5stickc/src/main.cpp.o
Compiling .pio/build/m5stickc/libe43/WiFi/WiFiUdp.cpp.o
In file included from src/main.cpp:31:
include/mqtt.h: In function 'void callbackSafety(byte*, unsigned int)':
include/mqtt.h:212:37: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:212:37: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
                                     SG_RELAY_ACTIVE_STATE
include/mqtt.h:218:36: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:218:36: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
                                    SG_RELAY_ACTIVE_STATE
src/main.cpp: In function 'void setup()':
src/main.cpp:314:35: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:314:35: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
                                   SG_RELAY_ACTIVE_STATE
Compiling .pio/build/m5stickc/lib178/ESPmDNS/ESPmDNS.cpp.o
*** [.pio/build/m5stickc/src/main.cpp.o] Error 1
Compiling .pio/build/m5stickc/liba6f/ArduinoOTA/ArduinoOTA.cpp.o
======================================================= [FAILED] Took 4.54 seconds =======================================================

Environment    Status    Duration
-------------  --------  ------------
m5stickc       FAILED    00:00:04.545
================================================= 1 failed, 0 succeeded in 00:00:04.545 =================================================

TinusDee, 2024-03-20:
Thnxs for all the updates, today I was able to flash the device. :)

raomin, 2024-03-21:
Great, sorry for these glitches. Enjoy!

vdiogo, 2024-03-21:
> Great, sorry for these glitches. Enjoy!

@raomin this issue still stands. See my last update
I don't think you should close this ticket

raomin, 2024-03-21:
@vdiogo I fixed the missing `SAFETY_RELAY_ACTIVE_STATE` and cross checked it's properly building.
If you still have an issue once the build is working, can you open a dedicated one please?

vdiogo, 2024-03-21:
> @vdiogo I fixed the missing `SAFETY_RELAY_ACTIVE_STATE` and cross checked it's properly building. If you still have an issue once the build is working, can you open a dedicated one please?

Apologies, all good! I used the build version from yesterday and was still getting the errors. Just downloaded, build an upload :)
Screen shows the "Setting up wifi..." message and now time to connect the stick to the pump

sdpeixoto, 2024-03-21:
@vdiogo Do you have your M5stickc Plus 2 device connected to a Windows machine through an USB port? If yes, have you changed anything in "upload_port" property or just left it as auto?

I was able to build the project successfully, but I'm not able to upload it to my device.

I'm getting below error:

-----------------------------------------
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
===================================== [FAILED] Took 83.80 seconds =====================================

Environment    Status    Duration
-------------  --------  ------------
m5stickcplus2  FAILED    00:01:23.799
================================ 1 failed, 0 succeeded in 00:01:23.799 ================================

 *  The terminal process "C:\Users\sdpeixoto\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'm5stickcplus2'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
-----------------------------------------

Any help will be much appreciated!

vdiogo, 2024-03-21:
> @vdiogo Do you have your M5stickc Plus 2 device connected to a Windows machine through an USB port? If yes, have you changed anything in "upload_port" property or just left it as auto?
> 
I'm using a Mac and the serial port was found properly

sdpeixoto, 2024-03-21:
Thanks @vdiogo for quick reply! Have you installed any driver to make M5stickc Plus 2 device being recognized?

sdpeixoto, 2024-03-21:
I found the reason for my M5stickc Plus 2 device not being recognized in Windows 11. It's because my USB cable is for charge only and it doesn't do for data as well.....After I connected another USB cable, just refresjhed the list of devices in Visual Studio, and here it is :)