raomin/ESPAltherma #410: Upload fails

markus786, 2024-03-14:
Hello!
I don't know what I'm doing wrong.
I've followed the instructions from start to finish step by step twice now, but every time I try to upload the code to my WEMOS D1 Mini ESP32, I get the error described below.
I've also tried selecting/including other environments and boards and even customizing the code a bit, but I can't get it fixed.
Is there anyone who knows this failure and can help me solve it?

```
Processing esp32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.5.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- PubSubClient @ 2.8.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- EEPROM @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
Building in release mode
Compiling .pio\build\esp32\src\main.cpp.o
Archiving .pio\build\esp32\liba0b\libPubSubClient.a
Archiving .pio\build\esp32\lib25f\libUpdate.a
Compiling .pio\build\esp32\lib807\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\esp32\lib807\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32\lib807\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32\lib807\WiFi\WiFiUdp.cpp.o
Compiling .pio\build\esp32\libb17\ESPmDNS\ESPmDNS.cpp.o
In file included from src/main.cpp:31:
include/mqtt.h: In function 'void readEEPROM()':
include/mqtt.h:64:84: error: 'PIN_THERM_ACTIVE_STATE' was not declared in this scope
     mqttSerial.printf("Restoring previous state: %s",(EEPROM.read(EEPROM_STATE) == PIN_THERM_ACTIVE_STATE)? "On":"Off" );
                                                                                    ^~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:64:84: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     mqttSerial.printf("Restoring previous state: %s",(EEPROM.read(EEPROM_STATE) == PIN_THERM_ACTIVE_STATE)? "On":"Off" );
                                                                                    ^~~~~~~~~~~~~~~~~~~~~~
                                                                                    SG_RELAY_ACTIVE_STATE
include/mqtt.h:69:32: error: 'PIN_THERM_ACTIVE_STATE' was not declared in this scope
     EEPROM.write(EEPROM_STATE,!PIN_THERM_ACTIVE_STATE);
                                ^~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:69:32: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     EEPROM.write(EEPROM_STATE,!PIN_THERM_ACTIVE_STATE);
                                ^~~~~~~~~~~~~~~~~~~~~~
                                SG_RELAY_ACTIVE_STATE
In file included from src/main.cpp:31:
include/mqtt.h: In function 'void callbackTherm(byte*, unsigned int)':
include/mqtt.h:135:30: error: 'PIN_THERM_ACTIVE_STATE' was not declared in this scope
     digitalWrite(PIN_THERM, !PIN_THERM_ACTIVE_STATE);
                              ^~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:135:30: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(PIN_THERM, !PIN_THERM_ACTIVE_STATE);
                              ^~~~~~~~~~~~~~~~~~~~~~
                              SG_RELAY_ACTIVE_STATE
include/mqtt.h:142:29: error: 'PIN_THERM_ACTIVE_STATE' was not declared in this scope
     digitalWrite(PIN_THERM, PIN_THERM_ACTIVE_STATE);
                             ^~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:142:29: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(PIN_THERM, PIN_THERM_ACTIVE_STATE);
                             ^~~~~~~~~~~~~~~~~~~~~~
                             SG_RELAY_ACTIVE_STATE
include/mqtt.h: In function 'void callbackSafety(byte*, unsigned int)':
include/mqtt.h:212:37: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:212:37: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
                                     SG_RELAY_ACTIVE_STATE
include/mqtt.h:218:36: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:218:36: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
                                    SG_RELAY_ACTIVE_STATE
src/main.cpp: In function 'void setup()':
src/main.cpp:314:35: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:314:35: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
                                   SG_RELAY_ACTIVE_STATE
Compiling .pio\build\esp32\lib94c\ArduinoOTA\ArduinoOTA.cpp.o
Compiling .pio\build\esp32\lib8b6\EEPROM\EEPROM.cpp.o
*** [.pio\build\esp32\src\main.cpp.o] Error 1
====================================================================================== [FAILED] Took 15.31 seconds ======================================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:15.312
================================================================================= 1 failed, 0 succeeded in 00:00:15.312 =================================================================================
```

Bobby5169, 2024-03-14:
I tried tonight and I have the same problem.
I'm totally newbie with esp.
Hope we will find a solution

raomin, 2024-03-15:
Hmm sorry guys I must have messed up with the latest pr... I'll fix it today

raomin, 2024-03-16:
Sorry guys I forgot to add PIN_THERM_ACTIVE_STATE on the public `setup.h`. Now it should work.

closed 2024-03-16 by raomin

Bobby5169, 2024-03-16:
Thanks for the work and the short delay for the resolution

markus786, 2024-03-17:
Hello raomin,

it still doesn't work for me.
Seems like you still need to include SAFETY_RELAY_ACTIVE_STATE and SG_RELAY_ACTIVE_STATE as you did it for PIN_THERM_ACTIVE_STATE !?

```
In file included from src/main.cpp:31:
include/mqtt.h: In function 'void callbackSafety(byte*, unsigned int)':
include/mqtt.h:212:37: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:212:37: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
                                     SG_RELAY_ACTIVE_STATE
include/mqtt.h:218:36: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:218:36: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
                                    SG_RELAY_ACTIVE_STATE
src/main.cpp: In function 'void setup()':
src/main.cpp:314:35: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:314:35: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
                                   SG_RELAY_ACTIVE_STATE
*** [.pio\build\esp32\src\main.cpp.o] Error 1
================= [FAILED] Took 31.43 seconds =================

Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:31.428
============ 1 failed, 0 succeeded in 00:00:31.428 ============ 
```

DanielBeck2, 2024-03-18:
Unfortunately I always get this error and I'm still a newbie in this area.
Is there an error in the file or am I doing something wrong?

In file included from src/main.cpp:31:
include/mqtt.h: In function 'void callbackSafety(byte*, unsigned int)':
include/mqtt.h:212:37: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:212:37: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
                                     SG_RELAY_ACTIVE_STATE
include/mqtt.h:218:36: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:218:36: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
                                    SG_RELAY_ACTIVE_STATE
src/main.cpp: In function 'void setup()':
src/main.cpp:314:35: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:314:35: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
                                   SG_RELAY_ACTIVE_STATE
Archiving .pio\build\m5stickc\libdb2\libWiFiClientSecure.a
Compiling .pio\build\m5stickc\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\m5stickc\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pio\build\m5stickc\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\m5stickc\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\m5stickc\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\m5stickc\FrameworkArduino\IPAddress.cpp.o
*** [.pio\build\m5stickc\src\main.cpp.o] Error 1
=========================== [FAILED] Took 10.74 seconds ===========================

Environment    Status    Duration
-------------  --------  ------------
m5stickc       FAILED    00:00:10.743
====================== 1 failed, 0 succeeded in 00:00:10.743 ======================

markus786, 2024-03-19:
@raomin Can we expect an answer from you in here? Although the issue was closed, after you had declared PIN_THERM_ACTIVE_STATE, the problem with SAFETY_RELAY_ACTIVE_STATE and SG_RELAY_ACTIVE_STATE seems not solved yet...

raomin, 2024-03-19:
I'm sorry, I missed this one yesterday but fixed it today. Can you cross check?

DanielBeck2, 2024-03-20:
> @raominKönnen The update works for me, but no values ​​are displayed despite selection, the only value is always AlthermaSensors online.

markus786, 2024-03-23:
Thanks @raomin! Now it works.

But unfortunately I just found another issue while I was setting up my system: #418