raomin/ESPAltherma #414: Build fails for M5stickc Plus

PiotrBeeScale, 2024-03-19:
1. I set the parameters in setup.h (WIFI, MQTT, Altherma model..) 
2. I have set the environment for M5stickplus 
Errors during compilation: 

Processing m5stickcplus (platform: espressif32; board: m5stick-c; framework: arduino)
---------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stick-c.html
PLATFORM: Espressif 32 (6.4.0) > M5Stick-C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny,
tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- M5StickCPlus @ 0.1.0
|-- PubSubClient @ 2.8.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- EEPROM @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
Building in release mode
Compiling .pio\build\m5stickcplus\src\main.cpp.o
In file included from src/main.cpp:25:
src/setup.h:124:2: warning: #warning "NO DEFINITION SELECTED: Please select your heat pump definition in /src/setup.h -- Using default." [-Wcpp]
 #warning "NO DEFINITION SELECTED: Please select your heat pump definition in /src/setup.h -- Using default."
  ^~~~~~~
In file included from src/main.cpp:31:
include/mqtt.h: In function 'void callbackSafety(byte*, unsigned int)':
include/mqtt.h:212:37: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:212:37: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
                                     SG_RELAY_ACTIVE_STATE
include/mqtt.h:218:36: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:218:36: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
                                    SG_RELAY_ACTIVE_STATE
src/main.cpp: In function 'void setup()':
src/main.cpp:314:35: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:314:35: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
                                   SG_RELAY_ACTIVE_STATE
Archiving .pio\build\m5stickcplus\lib394\libWiFi.a
Archiving .pio\build\m5stickcplus\lib272\libESPmDNS.a
Compiling .pio\build\m5stickcplus\lib247\ArduinoOTA\ArduinoOTA.cpp.o
Compiling .pio\build\m5stickcplus\lib8e1\EEPROM\EEPROM.cpp.o
Compiling .pio\build\m5stickcplus\libadc\WiFiClientSecure\WiFiClientSecure.cpp.o
Compiling .pio\build\m5stickcplus\libadc\WiFiClientSecure\esp_crt_bundle.c.o
*** [.pio\build\m5stickcplus\src\main.cpp.o] Error 1
Compiling .pio\build\m5stickcplus\libadc\WiFiClientSecure\ssl_client.cpp.o
================================ [FAILED] Took 13.62 seconds ================================

Environment    Status    Duration
-------------  --------  ------------
m5stickcplus   FAILED    00:00:13.621
=========================== 1 failed, 0 succeeded in 00:00:13.621 ===========================

 *  Działanie procesu terminalu „C:\Users\Piotr\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'm5stickcplus'” zostało zakończone z kodem zakończenia: 1. 
 *  Terminal zostanie ponownie użyty przez zadania. Naciśnij dowolny klawisz, aby go zamknąć. 

 *  Wykonywanie zadania: C:\Users\Piotr\.platformio\penv\Scripts\platformio.exe run --environment m5stickcplus 

Processing m5stickcplus (platform: espressif32; board: m5stick-c; framework: arduino)
---------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stick-c.html
PLATFORM: Espressif 32 (6.4.0) > M5Stick-C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny,
tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- M5StickCPlus @ 0.1.0
|-- PubSubClient @ 2.8.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
|-- EEPROM @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
Building in release mode
Compiling .pio\build\m5stickcplus\src\main.cpp.o
In file included from src/main.cpp:31:
include/mqtt.h: In function 'void callbackSafety(byte*, unsigned int)':
include/mqtt.h:212:37: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:212:37: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
                                     SG_RELAY_ACTIVE_STATE
include/mqtt.h:218:36: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
include/mqtt.h:218:36: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
     digitalWrite(SAFETY_RELAY_PIN, SAFETY_RELAY_ACTIVE_STATE);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
                                    SG_RELAY_ACTIVE_STATE
src/main.cpp: In function 'void setup()':
src/main.cpp:314:35: error: 'SAFETY_RELAY_ACTIVE_STATE' was not declared in this scope
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:314:35: note: suggested alternative: 'SG_RELAY_ACTIVE_STATE'
   digitalWrite(SAFETY_RELAY_PIN, !SAFETY_RELAY_ACTIVE_STATE);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
                                   SG_RELAY_ACTIVE_STATE
Archiving .pio\build\m5stickcplus\lib247\libArduinoOTA.a
Archiving .pio\build\m5stickcplus\lib8e1\libEEPROM.a
Archiving .pio\build\m5stickcplus\libadc\libWiFiClientSecure.a
Compiling .pio\build\m5stickcplus\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\m5stickcplus\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pio\build\m5stickcplus\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\m5stickcplus\FrameworkArduino\HWCDC.cpp.o
*** [.pio\build\m5stickcplus\src\main.cpp.o] Error 1
================================ [FAILED] Took 11.06 seconds ================================

Environment    Status    Duration
-------------  --------  ------------
m5stickcplus   FAILED    00:00:11.060
=========================== 1 failed, 0 succeeded in 00:00:11.060 ===========================

The version downloaded on November 28, 2023 compiles correctly.

raomin, 2024-03-19:
Sorry, I again forgot to align some new configuration element in the `setup.h`
Please update and try again.

PiotrBeeScale, 2024-03-19:
Thank you very much for your help, after updating setup.h everything works as it should.