raomin/ESPAltherma #418: Activating {0x60,5,203,1,-1,"Betrieb / Störung"} leads to an erroneous JSON.

markus786, 2024-03-23:
Activating 
`{0x60,5,203,1,-1,"Betrieb / Störung"}`
in the German
`ALTHERMA(LT_CA_CB_04-08KW).h`
leads to an erroneous JSON.

`2024-03-23 00:31:57.172 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: 
{"Aktuelle Betriebsart":"Fan Only","Thermostat EIN/AUS":"OFF",
...,
### "Betrieb / Störung":-,
...,
"FreeMem":"225752"}
`

raomin, 2024-03-23:
Can you paste the json message received from mqtt?

markus786, 2024-03-23:
When I activate the mentioned register I only get one single row with all the entries separated by commas, but not getting separated into attributes. As soon as I deactivate it, everything works again.

ck2510, 2025-02-14:
The umlaut ö could potentially be the problem try replacing it with oe.

raomin, 2026-10-09:
Fixed in the beta: a value without digits, like `-`, is now quoted, so the JSON stays valid. Thanks @markus786!

closed 2026-10-09 by raomin