raomin/ESPAltherma #421: Daikin heat pump indoor model: EKHBX016B3V3 "Error 0x15 0xEA returned from HP" and "Conv *** not avail" errors

sdpeixoto, 2024-03-25:
First of all, many thanks for your contribution on this project! It is a great help for all of us to control our Daikin heat pumps with a very reduced budget!!!

Daikin heat pump indoor model: EKHBX016B3V3
M5StackC Plus 2 Stick with the 8 pin X10A connector
ALTHERMA(LT_CA_CB_11-16KW).h definition

Home Assistant entities retrieved by me successfully from Daikin heat pump (see in attachment
[image]
)
 
Looking for “log” topic in MQTT explorer, I can see an error “**Error 0x15 0xEA returned from HP**” that is constantly being published….. Does thit relate to any configuration which is not compatible with my Daikin heat pump model?

I have uncommented all attributes from ALTHERMA(LT_CA_CB_11-16KW).h definition file except the ones that have in their description “Not in use”, and then have got in the JSON message references to “Conv *** not avail” as you can see below.
{
  ****"*Refrigerant type": "Conv 801 not avail.",****
  "Sensor Data Qty": 0,
  "INV compressor Qty": 0,
  "STD compressor Qty": 0,
  "Fan Data Qty": 0,
  "Expansion Valve Data Qty": 0,
  "4 Way Valve Data Qty": 0,
  "Crank Case Heater Qty": 0,
  "Solenoid valve Qty": 0,
  "Max. connectable indoor units": 0,
  "Connected Indoor Unit Qty": 0,
  "O/U MPU ID (xx)": 0,
  "O/U MPU ID (yy)": 0,
  "O/U capacity (kW)": 17.7,
  **"NextDataGrid": "Conv 995 not avail.",
  "In-Out separator": "Conv 998 not avail.",**
  "Operation Mode": "Fan Only",
  "Thermostat ON/OFF": "OFF",
  "Restart standby": "OFF",
  "Startup Control": "OFF",
  "Defrost Operation": "OFF",
  "Oil Return Operation": "OFF",
  "Pressure equalizing operation": "OFF",
  "Demand Signal": "OFF",
  "Low noise control": "OFF",
  "Error type": "Normal",
  "Error Code": " 0",
  "Target Evap. Temp.": 0,
  "Target Cond. Temp.": 0,
  "Discharge Temp. Drop": "OFF",
  **"Discharge Temp. Protection Retry Qty": "Conv 310 not avail.",**
  "Comp. INV Current Drop": "OFF",
  **"Comp. INV Current Protection Retry Qty": "Conv 311 not avail.",**
  "HP Drop Control": "OFF",
  **"HP Protection Retry Qty": "Conv 310 not avail.",**
  "LP Drop Control": "OFF",
  **"LP Protection Retry Qty": "Conv 311 not avail.",**
  "Fin Temp. Drop Control": "OFF",
  **"Fin Temp. Protection Retry Qty": "Conv 310 not avail.",**
  "Other Drop Control": "OFF",
  **"Not in use": "Conv 311 not avail.",**
  "R1T-Outdoor air temp.": 0,
  "O/U Heat Exch. Temp.": 0,
  "Discharge pipe temp.": 0,
  "Suction pipe temp.": 0,
  "Heat exchanger mid-temp.": 0,
  "Liquid pipe temp.(R6T)": 0,
  "Heat sink temp.": 0,
  "Pressure": 0,
  "Pressure(T)": -51.1813,
  "INV primary current (A)": 0,
  "INV secondary current (A)": 0,
  "Voltage (N-phase) (V)": -26880,
  "Brine Flow Switch": "OFF",
  "Brine inlet temp.": 0,
  "Brine outlet temp.": 0,
  "Refrig. temp. evap. In": 0,
  "Refrig. temp. evap.Out": 0,
  "INV frequency (rps)": 0,
  "Fan 1 (step)": 0,
  "Fan 2 (step)": 0,
  "Expansion valve 1 (pls)": 0,
  "Expansion valve 2 (pls)": 0,
  "4 Way Valve 1": "OFF",
  "4 Way Valve 2": "OFF",
  "4 Way Valve 3": "OFF",
  "4 Way Valve 4": "OFF",
  "4 Way Valve 5": "OFF",
  "Crank case heater 1": "OFF",
  "Crank case heater 2": "OFF",
  "Crank case heater 3": "OFF",
  "Crank case heater 4": "OFF",
  "SV1 (hot gas)": "OFF",
  "SV2 (BPH)": "OFF",
  "Solenoid Valve 3": "OFF",
  "Solenoid Valve 4": "OFF",
  "Solenoid Valve 5": "OFF",
  "Data Enable/Disable": "ON",
  "Indoor Unit Address": 0,
  "I/U operation mode": "DHW",
  "Freeze Protection": "OFF",
  "Silent Mode": "OFF",
  "Freeze Protection for water piping": "OFF",
  ******"Indoor Unit Code": "Conv 314 not avail.",
  "I/U capacity code": "Conv 219 not avail.",******
  "DHW setpoint": 45,
  "LW setpoint (main)": 50,
  "Water flow switch": "OFF",
  "Thermal Protector BUH": "OFF",
  "Thermal protector BSH": "ON",
  "Benefit kWh rate power supply": "OFF",
  "Solar input": "OFF",
  "Bivalent Operation": "OFF",
  "2way valve(On:Heat_Off:Cool)": "ON",
  "3way valve(On:DHW_Off:Space)": "OFF",
  "BSH": "OFF",
  "BUH Step1": "OFF",
  "BUH Step2": "OFF",
  "BPH": "OFF",
  "Water pump operation": "OFF",
  "Solar pump operation": "OFF",
  "Indoor Option Code": 43,
  "I/U Software ID (xx)": "{0:0}{1:0}",
  "I/U Software ID (yy)": "{0:1}{1:12}",
  "I/U EEPROM Ver.": 228,
  "Leaving water temp. before BUH (R1T)": 45,
  "Leaving water temp. after BUH (R2T)": 46.2,
  "Refrig. Temp. liquid side (R3T)": 29.9,
  "Inlet water temp.(R4T)": 44.1,
  "DHW tank temp. (R5T)": 45.1,
  "Indoor ambient temp. (R1T)": 24.5,
  "Ext. indoor ambient sensor (R6T)": 15.7,
  "M5BatV": "4.68V",
  "WifiRSSI": "-40dBm",
  "FreeMem": "204140"
}

I saw the following comment in README, and this relates to a conversion that was not implemented. Could you please confirm if this is the case? If yes, Is this something that can be implemented?
“If it says 'conv XXX not avail.' it is that I did not implement this specific conversion of value. If you need this value, create an issue and I'll implement it.”

sdpeixoto, 2024-03-25:
Serving as example for “*Refrigerant type” attribute:

{0x00,0,801,0,-1,"*Refrigerant type"}
Querying register 0x00... 
0x40 0x00 0x0e 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xb1 
"*Refrigerant type": "Conv 801 not avail."

sdpeixoto, 2024-03-29:
@raomin Do you believe that this is something that can be looked at by you? 
Thanks.

jesseklm, 2024-11-03:
“Refrigerant type” seems to be static, it has the following meanings:
801 --> R410A
802 --> R32
803 --> R22