raomin/ESPAltherma #427: Integrate multiple boards MQTT and HomeAssistant

merinsTDL, 2024-04-02:
Good day! 

I would like to monitor multiple Altherma units connected to the same HA and MQTT broker. 
Unfortunately, the current code hardcodes all MCUs connect to the same topic. 
I would like to introduce some variable that allows to set the unit name and then for each unit to split the data. 
Which would be the best solution to split them based on the unit?
1. To replace the MQTT topics based on unit name root i.e. <espaltherma>/LWT.
2. To add in the path the unit name i.e. espaltherma/<unit_name>/LWT.

KindeG, 2024-06-21:
Hello,
what I did is to change the topics - and also I recommend the 2. solution.
you should change it in a few files, basically:
setup.h (if onetopic is used...)
main.cpp (log...)
mqtt.h (/ATTR & /LWT & all the sensor publishing)

This way you can really do the things. I also put it into telegraf & grafana, that's why I suggest to have the unit name as well to feed the parser an easier recognition.

merinsTDL, 2024-07-02:
Thank you @KindeG for the suggestions! 
I will adjust it in this way!

merinsTDL, 2024-07-02:
Closing as resolved.

closed 2024-07-02 by merinsTDL