raomin/ESPAltherma #438: error: 'class WiFiClass' has no member named 'setSortMethod'

stosoorok, 2024-06-06:
Platformio fails, when I try to build latest version. 

I got two errors:

`src\main.cpp: In function 'void setup_wifi()':
src\main.cpp:218:8: error: 'class WiFiClass' has no member named 'setSortMethod'
   WiFi.setSortMethod(WIFI_CONNECT_AP_BY_SIGNAL);
        ^
src\main.cpp:219:8: error: 'class WiFiClass' has no member named 'setScanMethod'
   WiFi.setScanMethod(WIFI_ALL_CHANNEL_SCAN);
        ^
src\main.cpp: In function 'void setup()':
src\main.cpp:357:53: warning: format '%d' expects argument of type 'int', but argument 3 has type 'long unsigned int' [-Wformat=]
   Serial.printf("Wifi client timeout: %d\n", timeout);`

Any ideas?

[image]

[image]

raomin, 2024-06-11:
Probably some framework breaking change. I'll have a look.

stosoorok, 2024-06-16:
Thanks. BTW, where did you enable ESP wifi signal strength? I can't find it.

Rapanter01, 2024-06-16:
Hello, i have the same issue.
Did you found a solution?
Thanks

raomin, 2024-06-16:
That's weird, I get a successful build with the latest framework (6.7.0).

Can you try to force it: 
`platform = espressif32 @ 6.7.0`

or try to revert to an older one:
`platform = espressif32 @ 6.4.0`

Note: you can also remove the two lines, it should work just fine.

RE: Wifi signal, the wifi signal threshold was removed for, normally, getting always the best ssid.
At one point, we should move to `WiFiMulti.h`.

stosoorok, 2024-06-17:
Thnx @raomin I commented out these lines yesterday and I was able to compile it and upload. Scratch that wifi rssi. I was using your first version of ESPAltherma which has worked great. I see that many things have changed meanwhile.

I uploaded it via USB. I was not able to get it to upload via espota. What I am missing here?