raomin/ESPAltherma #442: Looking for the SG ports

dirkjanfaber, 2024-07-12:
First of all, what a superb project. I got connecting the ESP32 to my `EHVX08S18D6V` figured out by following the documentation. Thanks for developing this.

I would like to steer the device a bit too, mainly by enabling and disabling the SG ports, if possible. By looking at the labelling of the connections, I don't clearly see which ports that would be. On the right terminal I got the following connections:

[image]

On the left terminal, the labels are a bit harder to read because of all the cables in front of it, but they are labeled as follows:
[image]

Any idea what ports are linked to the SG ports?

dirkjanfaber, 2024-07-12:
Of course it could also be one of the other ports on the board: X42A, X32A, X85A, X38A or X33A. All suggestions are welcome. 

[image]

dirkjanfaber, 2024-07-12:
After some googling, it looks like the `S1S` contact might be the _benefit kWh rate power supply contact_. Which, if I understand correctly, seems equivalent to the SG1 contact. 

Other things I read is that only the LAN adapter will provide the extra ports.

luksa, 2024-11-16:
Yup, for older models, you need the [LAN adapter](https://www.daikin.eu/en_us/products/product.html/BRP069A61.html). But make sure to get the one with SmartGrid functionality.