raomin/ESPAltherma #494: AlthermaSensors is frequently unavailable for a short period of time

rogervdh, 2025-01-04:
Hi,

I have installed the latest version of ESPAltherma on my M5Stick Plus2. In the beginning it was quite stable. Sometime ESPAltherma was available for days without one interuptions and sometime sensors became unavailable for a short period of time. I expected that the WiFi connection wasn't stable. I moved the M5Stick a bit and the problem was fixed.

Since then the problem became (much) worse. At the moment sensors are frequently unavailable in Home Assistant (see the image below, all the grey small lines are interuptions). The M5Stick Plus2 is installed at +/- 1 meter distance from an (UTP wired) accesspoint, so the WiFi signal is strong (and the backbone connection via UTP is reliable). I have also checked the logging of the accesspoint and the M5Stick Plus2 isn't (de)registering.

[image]

I'm using the latest version of Home Assistant (core, OS, supervisor, etc), Mosquitto broker, etc. I don't have a clue what is causing this issue. I have already rebooted my Daikin Altherma and re-flashed my M5Stick Plus2, but without success.

antoniochiumiento, 2025-01-16:
I have the same problem.
Sometimes I see from my Unifi Connect that the m5stick plus 2 loses the wifi connection without any explanation (it is less than 3 meters from the Access Point)..
Other times it happens that despite being connected to the wifi it does not send any data to my Home Assistant.
I would be curious to know if there is any version of ESP that connects via Ethernet.

Jukeho, 2025-02-23:
Same here. Had dozend of timeouts a day.  Access Point is 1 meter away. I was trying everything I found on the Internet to get rid of this anoying behaviour. No success. 

Finally I changed the keep-alive in 
.pio/libdeps/m5stickcplus2/PubSubClient/src/PubSubClient.h
from 15 to 60

#define MQTT_KEEPALIVE 60

This fixed it completely. No timeouts since 2 weeks. No idea if this is the right way to change the parameter, but a least it is the only solution for me that worked.

Tiennetti, 2025-04-02:
Hi @Jukeho i tried your solution and it went well for a coule of weeks, now it started misbehaving like before, although with much less frequency
How is your ESPaltherma doing?

raomin, 2025-04-02:
There were some wifi issues at some point with different version of ESP-Arduino: if you have several access points with the same wifi name, the ESP might not connect to the strongest signal. This can cause some periodic loss. I tried to force it but it's still unreliable.
Do you have multiple AP with same names? Can you try to specify a name for the AP next to ESPAltherma?

Jukeho, 2025-04-02:
Since the change of this parameter I did not experience a single timeout. Before changing this parameter I observed what raomin describes: The ESP conected from time to time to the wrong AP that has a quite weak signal. I installes a new AP with a different SSID and connected the ESP to this AP. This helped reducing the timeouts, but the final solution was the change of the parameter. 

Looks to me that the combination of the 2 measures is the solution