raomin/ESPAltherma #496: Smart Grid mode "Unknown" or "Free Running" after short interruption or reboot HA

rogervdh, 2025-01-16:
When my M5Stick loses the WiFi connection or is unavailable of unknown for some other reason (sometimes only a few seconds), it looks like Home Assistant doesn't know the current smart grid mode anymore. The status becomes "Unknown" or "Free Running", while the actual smart grid mode is (for example) "Recommended on" (so SG1 is still closed and SG2 open). When an automation tries to change the SG mode to "Free Running" it isn't working, because Free Running is already the actual status in Home Assistant. What can I (or we) change to remain the correct SG status after a short (WiFi) interruption or reboot? Thanks in advance.

nicx, 2025-01-29:
@rogervdh same for me. my workaround is an automation triggered on HA start: When smart grid state is "unknown" set it to "free running". with that I have a defined state of the smart grid state.

Matssa56, 2025-02-18:
Same issue here and for me it's caused by the state not being either refreshed at startup, either the retain flag isn't turned on. This needs to be added in the MQTT firmware used for compiling.

rogervdh, 2025-02-23:
> Same issue here and for me it's caused by the state not being either refreshed at startup, either the retain flag isn't turned on. This needs to be added in the MQTT firmware used for compiling.

It seems like this request is covering the issues?
https://github.com/raomin/ESPAltherma/pull/457

Matssa56, 2025-02-23:
> It seems like this request is covering the issues? [#457](https://github.com/raomin/ESPAltherma/pull/457)

Yeah it seems like it but with a better approach that I suggested. However it needs to be merged

raomin, 2026-10-23:
@rogervdh Fixed in ESPAltherma 2 (beta.11): the real Smart Grid mode is published retained, so Home Assistant shows it again after a restart or a WiFi blip. A reboot of the ESP32 itself still comes back in Free Running, like the relays, on purpose. Closing, you've seen it live by now 🙂

closed 2026-10-23 by raomin