raomin/ESPAltherma #499: ESP32-S3 M5Stack ATOMS3 Lite - compatibility

HorvathTamasKristof, 2025-01-29:
Hello, 

I have bought (by accident) an ESP32-S3 instead of ESP32. 
[link to ESP32-S3 M5Stack ATOMS3 Lite](https://shop.m5stack.com/products/atoms3-lite-esp32s3-dev-kit)

During flashing by selecting the prepared _m5stickc_ profile (I tried almost all) I receive an error telling that this is an ESP32-S3 chip and not ESP32:
[image]

I would like to ask whether is there a way to get this chip working? 
If yes, what should I change in the code?

Thank you!
Tamas

ocsi01, 2025-02-02:
Actually I am just trying the same device as you have (as well as the M5NanoC6). I did not yet managed to make it work, but I managed to pass the issue what you are facing:

```
[env:m5stack-atoms3]
platform = espressif32@6.3.2
framework = arduino
board = m5stack-atoms3
upload_speed = 1500000
monitor_speed = 9600
upload_port = /dev/ttyACM0
build_flags = 
	-D CORE_DEBUG_LEVEL=5
	-D ARDUINO_USB_MODE=1
	-D ARDUINO_USB_CDC_ON_BOOT=1
	-D ARDUINO_M5Stick_Atom3S_Lite
lib_deps = 
	PubSubClient
	m5stack/M5Unified@^0.2.2
	m5stack/M5AtomS3@^1.0.1
	fastled/FastLED@^3.9.12
	Wire
```

It is not yet cleaned and I don´t exactly remember which settings was fixing this particular issue, however I can confirm, I've managed to build and deploy the code on the AtomS3-Lite. Wifi and MQTT works. ( I am having issues with reading the Tx/Rx right now.)

Note, I am using borth the M5Unified as well as the M5AtomS3 lib. It is only for flashing the lad for feedback. Since this has no screens, these are not needed. `ARDUINO_M5Stick_Atom3S_Lite` is used for the board specific pin numbers.

HorvathTamasKristof, 2025-02-03:
Thanks a lot for the proposal. 
I copied your lines and try to build the code. Visual studio says success but there is one warning. 

[image]

Do you think I can proceed with Upload with this or should be fixed?

Thanks!

ocsi01, 2025-02-05:
I guess this is not a critical issue. I would go for the upload. ( I can't check now if I have the same warning.) Sok sikert.

HorvathTamasKristof, 2025-02-06:
I uploaded and the unit is working fine.
Thanks a lot for giving the hint as solution. 

Köszi!

closed 2025-02-06 by HorvathTamasKristof

ocsi01, 2025-02-22:
If anyone wants to use the NanoC6 board ( which is amazing :D), you can find the board definition file here, since it's not yet in PlatformIO:
[m5stack-nanoc6.json](https://github.com/m5stack/M5UnitUnified/blob/main/boards/m5stack-nanoc6.json)
(Add this in your platfromIO project folder under `boards/m5stack-nanoc6.json` and you are ready to build.)

KalaNagHTD, 2025-05-04:
Mein AtomS3 läuft mit folgender Ergänzung in der platformio.ini:

`[env:m5stack-atomS3]
platform = espressif32
board = m5stack-atomS3
framework = arduino
upload_speed = 921600
monitor_speed = 115200

lib_deps = 
    m5stack/M5AtomS3
    m5stack/M5Unified
    knolleary/PubSubClient
	FastLED/FastLED `