raomin/ESPAltherma #505: TLS mqtt connection not working on ESP8226/nodemcuv2

Matssa56, 2025-03-04:
Hi!

I wanted to use the TLS encryption added with the commit 00e213a on my ESP8226 (D1 mini) and it doesn't work.

I tried without TLS and no issues. When adding the correct port and the TLS definition in setup.h, my D1 keeps rebooting.

Here are the logs that I can recover:
```
Restoring previous state: OnSetting up wifi...Connecting to <redacted>
............Connected. IP Address: <redacted>
Wifi client timeout: 5
Connecting to MQTT server: <redacted>:<redacted>
Attempting MQTT connection...failed, rc=-2, try again in 5 seconds
```

Here are some more logs (not sure if it can help) : 
```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffcd0 end: 3fffffc0 offset: 01a0
3ffffe70:  00c46a6f 00000000 00000000 40000001  
3ffffe80:  3fff0000 00000000 00000000 00000003
3ffffe90:  3fff16b4 3fff20a0 3fff20cc 4020d7d6
3ffffea0:  3fff53dc 3fff20a0 3fff20cc 4020dbf2
3ffffeb0:  3ffea7b7 00000008 3ffffec0 40208d60 <
3ffffec0:  00000000 4bc6a7f0 4c49ba5e 003a2398
3ffffed0:  00000000 00000000 4bc6a7f0 00000000
3ffffee0:  3fff16b4 3fff20a0 3fff15e4 4020ddeb  
3ffffef0:  3fff16b4 00002293 3fff15e4 4020c528
3fffff00:  00000001 3ffea7f5 00000001 402074ba
3fffff10:  3fff16b4 00002293 3fff11d4 4020231a
3fffff20:  00000001 3ffea7f5 00000000 00000000  
3fffff30:  00000000 3ffeafd6 3fffff40 3fff1804 <
3fffff40:  3fff1220 00001bd0 3fff11d4 3ffeafaf
3fffff50:  3fff1220 3fff11d4 3fff12c0 40202b5f  
3fffff60:  00000040 00000000 feefeffe feefeffe
3fffff70:  4020250c feefeffe 40211f0c 40211ef0
3fffff80:  000023f1 3fff16b4 feefeffe feefeffe
3fffff90:  feefeffe feefeffe feefeffe 3fff1804  
3fffffa0:  3fffdad0 00000000 3fff17f0 4020a7d8
3fffffb0:  feefeffe feefeffe 3ffea1a8 40101569
<<<stack<<<

last failed alloc call: 4020A554(16709)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v0006def0
~ld
```

Anyone managed to make it work on something else than an ESP32?

Nivalux, 2025-03-04:
I don't think I know the answer. I used a M5Stick_C_Plus.
Just to make sure, you did enable the `#define MQTT_ENCRYPTED`?

If you did and it still does not work, then I would try to see if there is a working example for your hardware (to test, if it works and see, if they are using the same libraries).
What I might also try is to send a GET request to a HTTPS Webserver of your choice, to test if you can get a response there.

If you want to stop the rebooting while debugging, you can disable it here:
https://github.com/raomin/ESPAltherma/blob/038f356b9c98cc6277059aecd80e801df0912963/include/mqtt.h#L121

Matssa56, 2025-03-04:
> I don't think I know the answer. I used a M5Stick_C_Plus. Just to make sure, you did enable the `#define MQTT_ENCRYPTED`?
> 
> If you did and it still does not work, then I would try to see if there is a working example for your hardware (to test, if it works and see, if they are using the same libraries). What I might also try is to send a GET request to a HTTPS Webserver of your choice, to test if you can get a response there.
> 
> If you want to stop the rebooting while debugging, you can disable it here:
> 
> [ESPAltherma/include/mqtt.h](https://github.com/raomin/ESPAltherma/blob/038f356b9c98cc6277059aecd80e801df0912963/include/mqtt.h#L121)
> 
> Line 121 in [038f356](/raomin/ESPAltherma/commit/038f356b9c98cc6277059aecd80e801df0912963)
> 
>  restart_board();

Thanks for your quick feedback. I did enable the define MQTT_ENCRYPTED. I tested just now an ESP32 board and it works fine, however I wanted to stick with my D1 mini since the Wifi seems to work better than my ESP32 (yeah a bit strange...).

I'll check if I can see something on the internet.

Nivalux, 2025-03-04:
I asked Chatgpt for an example program for the D1 to test the WiFi and noticed, that it used another WiFi header file. Maybe that is the reason.
Here is the example: 

```C
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "your_SSID";
const char* password = "your_PASSWORD";
const char* host = "example.com";  // Target URL

WiFiClientSecure client;

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
  Serial.println("Connected to WiFi");

  client.setInsecure();  // This is used to ignore the certificate (unsafe, not recommended for production applications)
  
  if (!client.connect(host, 443)) {
    Serial.println("Connection failed");
    return;
  }
  Serial.println("Connected to server");
  client.println("GET / HTTP/1.1");
  client.println("Host: example.com");
  client.println("Connection: close");
  client.println();
}

void loop() {
  // Empty loop
}
```

Matssa56, 2025-03-04:
> I asked Chatgpt for an example program for the D1 to test the WiFi and noticed, that it used another WiFi header file. Maybe that is the reason. Here is the example:
> 
> #include <ESP8266WiFi.h>
> #include <WiFiClientSecure.h>
> 
> const char* ssid = "your_SSID";
> const char* password = "your_PASSWORD";
> const char* host = "example.com";  // Target URL
> 
> WiFiClientSecure client;
> 
> void setup() {
>   Serial.begin(115200);
>   WiFi.begin(ssid, password);
>   while (WiFi.status() != WL_CONNECTED) {
>     delay(1000);
>     Serial.println("Connecting to WiFi...");
>   }
>   Serial.println("Connected to WiFi");
> 
>   client.setInsecure();  // This is used to ignore the certificate (unsafe, not recommended for production applications)
>   
>   if (!client.connect(host, 443)) {
>     Serial.println("Connection failed");
>     return;
>   }
>   Serial.println("Connected to server");
>   client.println("GET / HTTP/1.1");
>   client.println("Host: example.com");
>   client.println("Connection: close");
>   client.println();
> }
> 
> void loop() {
>   // Empty loop
> }

Yup saw that here : https://github.com/tsi-software/Secure_ESP8266_MQTT_poc/tree/master/top-level-components/secure_esp8266_mqtt_client#secure-esp8266-mqtt-client and here : https://github.com/knolleary/pubsubclient/pull/833/files

Gonna give it a try

Matssa56, 2025-03-04:
I saw somewhere that the clock must be set so that the certificate check can work. However I can't seem to find this check anywhere...

Matssa56, 2025-03-04:
Mmmh found something in WifiClientSecureBearSSL.h (called by ESP8226WiFi.h) : 
```
    // In cases when NTP is not used, app must set a time manually to check cert validity
    void setX509Time(time_t now) {
      _now = now;
    }
```
Maybe the time isn't correctly set on the esp, however not really sure why I don't have this issue on my esp32 ^^'

Nivalux, 2025-03-04:
For certificates to work you also need to embed all the root certificates that you want to trust. In this case, I set the http client to insecure, which basically skips the certificate validation. 
Not sure if you need the correct time, when you skip the check.
I have two different esp32 running and in neither have I set the time. Not sure if it even has a clock, that supports the current time instead of a looping runtime counter.

Matssa56, 2025-03-04:
> For certificates to work you also need to embed all the root certificates that you want to trust. In this case, I set the http client to insecure, which basically skips the certificate validation. [...]

Mmmh damn, I added the following code in main.cpp : 
```
 mqttSerial.print("Current time: ");
  time_t now = time(nullptr);
  struct tm timeinfo;
  gmtime_r(&now, &timeinfo);
  mqttSerial.print(asctime(&timeinfo));
```
Answer : Current time: Thu Jan  1 00:00:06 1970
Gonna test my esp32 to see. However if you disable the certificate check, then indeed the issue shouldn't come from here...

Matssa56, 2025-03-04:
@Nivalux just saw this : https://github.com/knolleary/pubsubclient/issues/185#issuecomment-1267791931

Especially the allow anonymous true. Since I'm using the Home Assistant MQTT broker addon, it doesn't support this. So my guess is that this is the reason why. IMO we either need to add the clock function, either change the broker... Gonna test it out tomorrow with a other broker with anonymous true.

Matssa56, 2025-03-11:
> I asked Chatgpt for an example program for the D1 to test the WiFi and noticed, that it used another WiFi header file. Maybe that is the reason. Here is the example:
>
> #include <ESP8266WiFi.h> [...]

Sooo I tried out some code with ChatGPT and I managed to make the TLS work : 

```
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <PubSubClient.h>  // Pour la gestion MQTT

// Remplacez par vos informations Wi-Fi
const char* ssid = "XXXX";
const char* password = "XXXX";

// Adresse du serveur MQTT
const char* mqtt_server = "XXXX";  // Remplace par l'adresse de ton serveur MQTT
const int mqtt_port = XXXX;                    // Port sécurisé MQTT (SSL/TLS)
const char* mqtt_user = "XXXX";            // Utilisateur pour la connexion MQTT (si nécessaire)
const char* mqtt_password = "XXXX"; // Mot de passe pour la connexion MQTT (si nécessaire)

WiFiClientSecure espClient;
PubSubClient client(espClient);

// Fonction de gestion de reconnexion MQTT
void reconnectMQTT() {
  while (!client.connected()) {
    Serial.print("Tentative de reconnexion MQTT...");
    
    if (client.connect("ESP8266Client", mqtt_user, mqtt_password)) {
      Serial.println("Reconnecté au serveur MQTT");
      // Une fois connecté, publier un message sur un topic
      client.publish("test/topic", "Hello World");
    } else {
      Serial.print("Échec, code erreur: ");
      Serial.print(client.state());
      delay(5000);  // Attente de 5 secondes avant de réessayer
    }
  }
}

// Callback lorsque des messages sont reçus
void mqttCallback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message reçu sur le topic: ");
  Serial.println(topic);

  String message = "";
  for (int i = 0; i < length; i++) {
    message += (char)payload[i];
  }
  Serial.println("Message: " + message);
}

void setup() {
  // Initialisation de la communication série
  Serial.begin(115200);

  Serial.print("ESP Board MAC Address:  ");
  Serial.println(WiFi.macAddress());
  
  // Connexion au Wi-Fi
  Serial.println();
  Serial.print("Connexion au Wi-Fi ");
  Serial.println(ssid);

  // Connexion au réseau Wi-Fi
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(".");
  }
  Serial.println("Connecté au Wi-Fi");

  Serial.print("Adresse IP: ");
  Serial.println(WiFi.localIP());

  espClient.setInsecure();

  // Initialisation de la connexion MQTT
  client.setServer(mqtt_server, mqtt_port); // Définir le serveur MQTT
  client.setCallback(mqttCallback); // Optionnel : callback lorsque des messages sont reçus

  // Ajouter le certificat SSL au client
  //espClient.setTrustAnchors(new X509List(mqtt_cert));

  // Tentative de connexion au serveur MQTT
  while (!client.connected()) {
    Serial.print("Tentative de connexion MQTT...");
    
    // Tentative de connexion avec un identifiant unique
    if (client.connect("ESP8266Client", mqtt_user, mqtt_password)) {
      Serial.println("Connecté au serveur MQTT");

      // Une fois connecté, publier un message sur un topic
      client.publish("test/topic", "Hello World");  // Publie "Hello World" sur le topic "test/topic"

    } else {
      Serial.print("Échec, code erreur: ");
      Serial.print(client.state());
      delay(5000);  // Attente de 5 secondes avant de réessayer
    }
  }
}

void loop() {
  // Gérer la connexion au serveur MQTT
  if (!client.connected()) {
    reconnectMQTT();
  }
  
  // Maintenir la connexion et traiter les messages entrants
  client.loop();
}
```

So I now have a base in order to see why the espaltherma code doesn't work, gonna see step by step (maybe there is a timeout somewhere that craps it out or somthing)

raomin, 2026-10-23:
@Matssa56 Sorry for the delay. The ESP8266 stays on ESPAltherma 1.x without TLS; ESPAltherma 2 (ESP32 only) has TLS working. Closing.

closed 2026-10-23 by raomin