raomin/ESPAltherma #511: Stalled values – Outdoor Temperature (0x20, 0, 105, 2, 1) – but: ONECTA is correct

dexamenos, 2025-04-11:
Hello, 

I am running into another "stalled value" problem (see #75).

My HP is Altherma 3 R ECH2O / Model EHSX04P50EF

I am using Daikin Onecta integration (https://github.com/jwillemsen/daikin_onecta) from @jwillemsen in Home Assistant. 

**And** I am using an ESP32 with ESPAltherma from @raomin. def file `Altherma(ERGA E EHSH-X P30-50 E_EF series 04-08kW-ECH2O).h`

In the following screenshot, top graph is the Outdoor Temperature, as reported by ONECTA API (`sensor.altherma_climatecontrol_outdoor_temperature`). Update frequency is once per 20 Minutes. However, I am using a custom python script to fold the Outdoor temperature as reported by ESPAltherma (`(0x20, 0, 105, 2, 1)`) onto the ONECTA integration entity. I am using MQTT to ATTR. (Yellow curve is weather forecast.)

Bottom graph is actual power consumption from the electricity counter just for the heat pump.

[image]

**Outdoor temperature stopped to be accurate two days ago!** What happened? I stopped the heating, because it is getting warmer now, so just hot water is running. What can you see? The ESPAltherma outdoor temperature stalls, as soon as outdoor unit is turned off, however, ONECTA still knows the correct value (those 20 minute peaks). When hot water is beeing prepared, ESPAltherma sensor is back to the correct value.

Any ideas?

mdrayner, 2025-06-17:
I have the same issue.. 0x61, 14, 105, 2, 1, reads 0 as I do not have an external temp sensor installed and 0x20, 0, 105, 2, 1 seems to get stuck unless the pump is running (has to be actually running, not just 'on' which in the summer is only about 45 mins during water heating)
The Daikin app still shows the correct readings so it must be accessible somewhere

ppiotrek90, 2025-10-07:
I have a similar issue. In my case, when the heating is off, the temprature is not updated. As soon as the heating is on, the temperature readings are more often updated.

I'm using this definition: **{0x20,0,105,2,1,"R1T-Outdoor air temp."},**

[image]

ppiotrek90, 2025-12-03:
> As far as I know when the Altherma is not running not all sensors are updated to save power.

Could be the reason but Altherma is actually on. Specifically, while DHW is running, it’s solely the space heating setting affecting the temperature. It’s simply strange.