raomin/ESPAltherma #513: ESP32 Wroom32 unable to query registers?

henrykuijpers, 2025-04-21:
In `src/setup.h` I entered my WiFi SSID & password & added my MQTT details. They're correct.

Then uncommented:
```
#include "def/Altherma(ERGA D EHV-EHB-EHVZ DA series 04-08kW).h"
```
(I have the Daikin Altherma ERGA08EVH7 + EHBX08E6V)

I connected 2-TX to RX2 (16) & 3-RX to TX2 (17). I connected 5-GND to the GND on the ESP32, the one next to 3V3.

This is the ESP32 board I'm using:

[image]

I get the following output on the device monitor using PlatformIO. It looks like something is very wrong. Does anyone recognize this error and is there a way to fix this?

```
xxx@NL01PA-NB039 ESPAltherma % pio device monitor
--- Terminal on /dev/cu.usbserial-0001 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Restoring previous state: OffSetting up wifi...Connecting to I am Groot
......Connected. IP Address: 192.168.1.79
Wifi client timeout: 1000
Connecting to MQTT server: 192.168.1.2:1883
Attempting MQTT connection...connected!
OK!
Adding registry 0x10 to be queried.
ESPAltherma started!Querying register 0x10... Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  0 register dump:
PC      : 0x400837d2  PS      : 0x00060130  A0      : 0x800835c0  A1      : 0x3ffcd3b0  
A2      : 0x000000da  A3      : 0x00001800  A4      : 0x40083f37  A5      : 0x00001000  
A6      : 0x00000000  A7      : 0x3ffb03e0  A8      : 0x00000000  A9      : 0x3ffcd380  
A10     : 0x00060034  A11     : 0x00000002  A12     : 0x00000000  A13     : 0x00060723  
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x00000000  EXCCAUSE: 0x00000009  
EXCVADDR: 0x40083f53  LBEG    : 0x40089910  LEND    : 0x40089926  LCOUNT  : 0xffffffff  

Backtrace: 0x400837cf:0x3ffcd3b0 0x400835bd:0x3ffcd3e0 0x400835e9:0x3ffcd400 0x40091521:0x3ffcd420 0x40084b61:0x3ffcd440 0x4017832d:0x3ffcd460 0x4017afd0:0x3ffcd490 0x4017a592:0x3ffcd4c0 0x4017c66d:0x3ffcd510 0x4017c737:0x3ffcd560 0x40177323:0x3ffcd5a0 0x4008e842:0x3ffcd5d0

ELF file SHA256: 0317033e2b01318a

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15468
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3180
entry 0x400805b8
Restoring previous state: OffSetting up wifi...Connecting to I am Groot
......Connected. IP Address: 192.168.1.79
Wifi client timeout: 1000
Connecting to MQTT server: 192.168.1.2:1883
Attempting MQTT connection...connected!
OK!
Adding registry 0x10 to be queried.
ESPAltherma started!Querying register 0x10... Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  0 register dump:
PC      : 0x400837d2  PS      : 0x00060f30  A0      : 0x800835c0  A1      : 0x3ffbad40  
A2      : 0x00000008  A3      : 0x00001800  A4      : 0x40083f37  A5      : 0x00001000  
A6      : 0x00000000  A7      : 0x3ffb03e0  A8      : 0x00000000  A9      : 0x3ffbad20  
A10     : 0x00060034  A11     : 0x00060723  A12     : 0x00060720  A13     : 0x3f40a510  
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x00000018  EXCCAUSE: 0x00000009  
EXCVADDR: 0x40083f53  LBEG    : 0x40084919  LEND    : 0x40084921  LCOUNT  : 0x00000027  

Backtrace: 0x400837cf:0x3ffbad40 0x400835bd:0x3ffbad70 0x400835e9:0x3ffbad90 0x40091521:0x3ffbadb0 0x40084b61:0x3ffbadd0 0x4014491f:0x3ffbadf0 0x4014d4db:0x3ffbae10 0x4014aad9:0x3ffbae30 0x400e6f15:0x3ffbae50 0x4008e842:0x3ffbae80

ELF file SHA256: 0317033e2b01318a

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15468
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3180
entry 0x400805b8
Restoring previous state: OffSetting up wifi...Connecting to I am Groot
......Connected. IP Address: 192.168.1.79
Wifi client timeout: 1000
Connecting to MQTT server: 192.168.1.2:1883
Attempting MQTT connection...connected!
OK!
Adding registry 0x10 to be queried.
ESPAltherma started!Querying register 0x10... ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15468
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3180
entry 0x400805b8
Restoring previous state: OffSetting up wifi...Connecting to I am Groot
......Connected. IP Address: 192.168.1.79
Wifi client timeout: 1000
Connecting to MQTT server: 192.168.1.2:1883
Attempting MQTT connection...connected!
OK!
Adding registry 0x10 to be queried.
ESPAltherma started!Querying register 0x10... Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  0 register dump:
PC      : 0x400837d2  PS      : 0x00060130  A0      : 0x800835c0  A1      : 0x3ffcd3b0  
A2      : 0x000000da  A3      : 0x00001800  A4      : 0x40083f37  A5      : 0x00001000  
A6      : 0x00000000  A7      : 0x3ffb03e0  A8      : 0x00000000  A9      : 0x3ffcd380  
A10     : 0x00060034  A11     : 0x00000002  A12     : 0x00000000  A13     : 0x00060723  
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x00000000  EXCCAUSE: 0x00000009  
EXCVADDR: 0x40083f53  LBEG    : 0x40089910  LEND    : 0x40089926  LCOUNT  : 0xffffffff  

Backtrace: 0x400837cf:0x3ffcd3b0 0x400835bd:0x3ffcd3e0 0x400835e9:0x3ffcd400 0x40091521:0x3ffcd420 0x40084b61:0x3ffcd440 0x4017832d:0x3ffcd460 0x4017afd0:0x3ffcd490 0x4017a592:0x3ffcd4c0 0x4017c66d:0x3ffcd510 0x4017c737:0x3ffcd560 0x40177323:0x3ffcd5a0 0x4008e842:0x3ffcd5d0

ELF file SHA256: 0317033e2b01318a
```

henrykuijpers, 2025-04-21:
Apparently during my clicking around trying to find information, this showed up in the MQTT monitoring that I set up:
```
espaltherma/LWT Offline
espaltherma/LWT Online
espaltherma/log OK!
espaltherma/log 

espaltherma/log Adding registry 0x10 to be queried.

espaltherma/log ESPAltherma started!
espaltherma/LWT Offline
```

But I have no idea what the device monitor was saying at that point.

It really looks like something is very wrong here.

henrykuijpers, 2025-04-21:
This is exactly the board I'm using: https://www.circuitstate.com/pinouts/doit-esp32-devkit-v1-wifi-development-board-pinout-diagram-and-reference/

raomin, 2026-10-23:
@henrykuijpers It's been a while. ESPAltherma 2 installs from the browser, sets the pins for your board and detects the unit, which tells a wiring problem from a definition one. Closing with the 2.0 release; if the DevKit still gets no answer on v2, the Diagnostics page will say why.

closed 2026-10-23 by raomin